Brandawg93/PeaNUT #6: [Feature Request]: Charts for other metrics

armond1978, 2023-10-24:
### Enhancement of issue

It would be great if we can monitor the load percentage, battery charge percentage and the battery runtime over time. This will be particularly helpful if there is a power outage and you want to understand how the UPS battery is performing and how you need to react.

### Solution

Just like the Input/Output Voltage chart, that would be very help if we could have charts for the load percentage, battery charge percentage and the battery runtime.

Brandawg93, 2023-10-24:
I've actually spent a lot of time trying to get that working. Turns out, it not very easy / user friendly to have a time based DB set up in docker. 

I've been weighing the pros and cons of forcing the user to run their own influxdb instance versus baking it into the image. 

For now, you can use the new API to periodically send the data to influx and set up a grafana dashboard if you want. 

But this is definitely on my todo list.

armond1978, 2023-10-26:
I'm not sure I understand why you need to capture the stats that frequently. Even once a minute can do the job. You're just going to monitor the overall trends. Every second will put a big pressure on the UPS network card as well and really not necessary in my opinion. But maybe I'm missing something here...

Brandawg93, 2023-10-26:
The UPS is already being polled by the NUT server. My plan would be to make it use the poll frequency already set up by the NUT server itself.

armond1978, 2023-10-26:
I was not aware of the NUT server internals. In that case, you should be good. Thanks for the explanation.

Brandawg93, 2024-06-17:
I've added a very rudimentary example of connecting the API to InfluxDB. This will definitely help until I can get around to adding this to the app itself.

https://github.com/Brandawg93/PeaNUT/tree/main/examples/persistant-data

Brandawg93, 2024-10-22:
With the latest version, you can use Grafana to create graphs for all data available within PeaNUT. Check out the [example](https://github.com/Brandawg93/PeaNUT/blob/main/examples/persistant-data/docker-compose.yml) for more info.

closed 2024-10-22 by Brandawg93

freitasm, 2025-05-08:
Hi @Brandawg93 

The example goes 404. I have an InfluxDB bucket configured on PeaNUT but I don't see any charts - something like https://raw.githubusercontent.com/Brandawg93/PeaNUT/main/images/charts.png - or is this using Grafana?

Brandawg93, 2025-05-08:
That's not grafana, just the UI I created within PeaNUT. You may have a configuration error within grafana, or your ups may not have the required fields. You could also try the user-create chart. It's pretty awesome.

https://github.com/zephyr325/Grafana-for-PeaNUT/tree/main

freitasm, 2025-05-09:
Thanks. Just interested in the PeaNUT charts really. 

Assuming the UPS has the data (WinNUT client on a Windows desktop shows that data), do the PeaNUT charts require the InfluxDB to work at all or is it using local storage?

Brandawg93, 2025-05-09:
No, influx isn’t needed to make PeaNUT work, only if you want persistent storage.