Brandawg93/PeaNUT #8: [Feature Request]: Authentication support

opticsthewebsite, 2023-12-10:
### Enhancement of issue

There is currently no way to use PeaNUT with NUT servers that require authentication

### Solution

Add the ability to access NUT servers that require authentication. Likely, this would be accomplished using two docker environment variables.

Brandawg93, 2023-12-10:
Could you provide me with a sample of your configuration files that make the NUT server require a username and password?

I've tried adding this before but can't seem to force my NUT server to require a login.

opticsthewebsite, 2023-12-10:
Unfortunately I can't be of any use there. My UPS is connected to a Synology Diskstation, which uses NUT under the hood for its [UPS monitoring](https://kb.synology.com/en-ph/DSM/help/DSM/AdminCenter/system_hardware_ups?version=7). Synology uses a hard coded NUT login/password. It's my understanding that QNAP is the same. I am able to use the NUT integration in Home Assistant as well as the NUT Telegraf plugin, since those support specifying a login/password. Sorry I can't be of more help.

(TBF Synology annoys me to no end, this just being one of countless things. I experimented with connecting my UPS to a RPi and running NUT on that, but at the end of the day, enjoyed the "Single Pane of Glass" experience that Synology provides a little more)

Brandawg93, 2023-12-10:
I actually have mine connected to a Synology as well. I just don't see anywhere to specify a username/password.
[image]

When I connect peanut to it, it just shows up. Does your configuration look different?

opticsthewebsite, 2023-12-10:
Well... uhh.. I tried it without specifying a login/password and it worked. That was... unexpected. I am fairly certain I tried setting up the [[inputs.upsd]] Telegraf plugin a while ago without specifying a login/password and it didn't work until I supplied it. Same with the Home Assistant NUT integration. (BTW, the hardcoded login/pass for Synology NUT is "monuser"/"secret") Maybe something changed when I upgraded to DSM 7.2. Or maybe you figured out a way to [hack the Gibson](https://pbs.twimg.com/media/F6E9sPfXcAAsDTY?format=jpg&name=small). I just assumed PeaNUT would require it as well without even trying. Either way, it works. Sorry to waste your time 😬

closed 2023-12-10 by opticsthewebsite

Brandawg93, 2023-12-10:
No worries! IIRC, getting information about a UPS doesn't require a login, but changing settings does. And since peanut only gets information currently, it isn't required. I've experimented with adding the ability to change settings, and I assume I will need to add this functionality eventually anyways.