Brandawg93/PeaNUT #12: [Bug]: API response for a system with two UPS seems to be mixing the two.

dimatx, 2023-12-25:
### Describe the bug

API response showing info about first UPS in the node for the second UPS. The second UPS is a Smart-UPS C 1500, but it shows a few lines referencing the first one: UPS-APC-Back-UPS-ES-750G

### Debug Output

[
    {
        "battery.charge": "100",
        "battery.charge.low": "10",
        "battery.charge.warning": "50",
        "battery.date": "2001/09/25",
        "battery.mfr.date": "2012/03/05",
        "battery.runtime": "35550",
        "battery.runtime.low": "120",
        "battery.type": "PbAc",
        "battery.voltage": "13.6",
        "battery.voltage.nominal": "12.0",
        "device.mfr": "American Power Conversion",
        "device.model": "Back-UPS ES 750G",
        "device.serial": "4B1210P05924  ",
        "device.type": "ups",
        "driver.name": "usbhid-ups",
        "driver.parameter.pollfreq": "30",
        "driver.parameter.pollinterval": "1",
        "driver.parameter.port": "auto",
        "driver.parameter.productid": "0002",
        "driver.parameter.serial": "4B1210P05924",
        "driver.parameter.synchronous": "no",
        "driver.parameter.vendorid": "051D",
        "driver.version": "2.7.4",
        "driver.version.data": "APC HID 0.96",
        "driver.version.internal": "0.41",
        "input.sensitivity": "medium",
        "input.transfer.high": "139",
        "input.transfer.low": "92",
        "input.transfer.reason": "input voltage out of range",
        "input.voltage": "121.0",
        "input.voltage.nominal": "120",
        "ups.beeper.status": "enabled",
        "ups.delay.shutdown": "20",
        "ups.firmware": "841.I4 .D",
        "ups.firmware.aux": "I4 ",
        "ups.load": "0",
        "ups.mfr": "American Power Conversion",
        "ups.mfr.date": "2012/03/05",
        "ups.model": "Back-UPS ES 750G",
        "ups.productid": "0002",
        "ups.realpower.nominal": "450",
        "ups.serial": "4B1210P05924  ",
        "ups.status": "OL",
        "ups.test.result": "No test initiated",
        "ups.timer.reboot": "0",
        "ups.timer.shutdown": "-1",
        "ups.vendorid": "051d",
        "BEGIN": "LIST VAR ups",
        "VAR": "ups device.serial 3S1623X01067"
    },
    {
        "BEGIN": "LIST VAR UPS-APC-Back-UPS-ES-750G",
        "VAR": "UPS-APC-Back-UPS-ES-750G ups.vendorid 051d",
        "END": "LIST VAR UPS-APC-Back-UPS-ES-750G",
        "": "",
        "battery.charge": "100",
        "battery.charge.low": "10",
        "battery.charge.warning": "50",
        "battery.runtime": "2100",
        "battery.runtime.low": "120",
        "battery.type": "PbAc",
        "battery.voltage": "27.3",
        "battery.voltage.nominal": "24.0",
        "device.mfr": "American Power Conversion ",
        "device.model": "Smart-UPS C 1500",
        "device.serial": "3S1623X01067  ",
        "device.type": "ups",
        "driver.name": "usbhid-ups",
        "driver.parameter.pollfreq": "30",
        "driver.parameter.pollinterval": "1",
        "driver.parameter.port": "auto",
        "driver.parameter.productid": "0003",
        "driver.parameter.synchronous": "no",
        "driver.parameter.vendorid": "051D",
        "driver.version": "2.7.4",
        "driver.version.data": "APC HID 0.96",
        "driver.version.internal": "0.41",
        "ups.beeper.status": "disabled",
        "ups.delay.shutdown": "20",
        "ups.firmware": "UPS 10.0 / ID=1005",
        "ups.mfr": "American Power Conversion ",
        "ups.mfr.date": "2016/06/07",
        "ups.model": "Smart-UPS C 1500",
        "ups.productid": "0003",
        "ups.serial": "3S1623X01067  ",
        "ups.status": "OL",
        "ups.timer.reboot": "-1",
        "ups.timer.shutdown": "-1",
        "ups.vendorid": "051d"
    }
]

### Steps to reproduce

1. 

### Device Type

Intel NUC

### UPS Device

usbhid-ups

Brandawg93, 2023-12-26:
Yep. Seems like a parsing error. I just published a new test version. You can try it by adding the `:test` tag. (brandawg93/peanut:test). Try it and let me know if that fixes it for you!

dimatx, 2023-12-27:
Just tried with :test and still seems like an issue persists. Here's a gif of me refreshing the page (see line 50 in the gif)... And also screencap of the extra params visible in the UI.

[image]

[image]

Brandawg93, 2023-12-27:
Can you tell me what the nut upsd server says the name of the ups is?

You should be able to telnet into the server and run `LIST UPS`.

Brandawg93, 2023-12-28:
Ok. I've reworked the parsing algorithm and published `:test` again. Let me know if it works now.

dimatx, 2023-12-28:
Issue appears to persist:

[image]

Here's what I get via running 'upsc {name}@localhost' on the host (RPi)

```
pi@rpi0:~ $ upsc UPS-APC-Back-UPS-ES-750G@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.date: 2001/09/25
battery.mfr.date: 2012/03/05
battery.runtime: 35550
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 13.6
battery.voltage.nominal: 12.0
device.mfr: American Power Conversion
device.model: Back-UPS ES 750G
device.serial: 4B1210P05924  
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.productid: 0002
driver.parameter.serial: 4B1210P05924
driver.parameter.synchronous: no
driver.parameter.vendorid: 051D
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
input.sensitivity: medium
input.transfer.high: 139
input.transfer.low: 92
input.transfer.reason: input voltage out of range
input.voltage: 120.0
input.voltage.nominal: 120
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 841.I4 .D
ups.firmware.aux: I4 
ups.load: 0
ups.mfr: American Power Conversion
ups.mfr.date: 2012/03/05
ups.model: Back-UPS ES 750G
ups.productid: 0002
ups.realpower.nominal: 450
ups.serial: 4B1210P05924  
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

```
pi@rpi0:~ $ upsc ups@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 2100
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.3
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion 
device.model: Smart-UPS C 1500
device.serial: 3S1623X01067  
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.productid: 0003
driver.parameter.synchronous: no
driver.parameter.vendorid: 051D
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.firmware: UPS 10.0 / ID=1005
ups.mfr: American Power Conversion 
ups.mfr.date: 2016/06/07
ups.model: Smart-UPS C 1500
ups.productid: 0003
ups.serial: 3S1623X01067  
ups.status: OL
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d
```

Brandawg93, 2023-12-28:
It's really odd that it changes. The parsing should be the same every time and it explicitly removes `VAR` from every line. 

Could you try using telnet directly and see what you get?

And thanks again for troubleshooting this with me!

Brandawg93, 2023-12-28:
Ok. I've pushed another test. This one replaces ALL occurrences of `VAR`. Let's see what that does.

dimatx, 2023-12-28:
Thanks for attempting the fix! Hmm... Same issue...  Can you elaborate on where you'd like me to telnet? I use SSH to connect to the raspberry pi running the NUT server. Happy to run whatever commands would be helpful.

In case it helps, this is the output from `pi@rpi0:~ $ sudo cat /etc/nut/ups.conf`

```
[UPS-APC-Back-UPS-ES-750G]
    driver = usbhid-ups
    port = auto
    desc = "APC Back-UPS ES 750G"
    vendorid = 051D
    productid = 0002
    serial = 4B1210P05924

[ups]
    driver = usbhid-ups
    port = auto
    desc = "APC Smart-UPS C 1500"
    vendorid = 051D
    productid = 0003
```

Greetings from just north of Boston, by the way.

Brandawg93, 2023-12-28:
Greetings fellow Bostonian!

You can telnet directly into the NUT server by doing something like `telnet localhost 3493` replacing the relevant hostname and port. This is how [peanut gets its information](https://github.com/Brandawg93/PeaNUT/blob/main/src/server/nut.ts).

You could also clone this repo and add a few `console.log`s as well.

My theory is that you will get the same changing info directly from the server. If that's the case, I'm not sure what else could be done to fix it.

dimatx, 2023-12-28:
I'll see if I can figure that out... And just as a data point, for what it's worth, I am also using this web UI, and it does not experience the same issue. Perhaps the method is different there... https://github.com/teknologist/docker-webnut

Brandawg93, 2023-12-28:
Yep. That uses [nut2](https://github.com/mezz64/python-nut2/blob/master/pynut2/nut2.py) under the hood. Ok. I've modified the parsing to be EXACTLY like it and made another test. 🤞

Brandawg93, 2023-12-29:
I know what it is! The peanut was requesting ups data asynchronously so the two devices were mixing lines. New test is ready!

dimatx, 2023-12-29:
Getting a 500 with some errors in the logs of the container.

```
   ▲ Next.js 14.0.4
   - Local:        http://7a6fd986e243:8080
   - Network:      http://192.168.112.2:8080
 ✓ Ready in 118ms
 ⨯ Error: Invalid response
    at n.getData (/.next/server/app/api/v1/devices/route.js:3:10)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async l (/.next/server/app/api/v1/devices/route.js:1:755)
    at async /node_modules/.pnpm/next@14.0.4_@babel+core@7.23.6_react-dom@18.2.0_react@18.2.0/node_modules/next/dist/compiled/next-server/app-route.runtime.prod.js:6:41960
 ⨯ Error: Invalid response
    at n.getData (/.next/server/app/api/v1/devices/route.js:3:10)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async l (/.next/server/app/api/v1/devices/route.js:1:755)
    at async /node_modules/.pnpm/next@14.0.4_@babel+core@7.23.6_react-dom@18.2.0_react@18.2.0/node_modules/next/dist/compiled/next-server/app-route.runtime.prod.js:6:41960
 ⨯ Error: Invalid response
    at n.getData (/.next/server/app/api/v1/devices/route.js:3:10)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async l (/.next/server/app/api/v1/devices/route.js:1:755)
    at async /node_modules/.pnpm/next@14.0.4_@babel+core@7.23.6_react-dom@18.2.0_react@18.2.0/node_modules/next/dist/compiled/next-server/app-route.runtime.prod.js:6:41960
 ⨯ Error: Invalid response
    at n.getData (/.next/server/app/api/v1/devices/route.js:3:10)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async l (/.next/server/app/api/v1/devices/route.js:1:755)
    at async /node_modules/.pnpm/next@14.0.4_@babel+core@7.23.6_react-dom@18.2.0_react@18.2.0/node_modules/next/dist/compiled/next-server/app-route.runtime.prod.js:6:41960
 ⨯ Error: Invalid response
    at n.getData (/.next/server/app/api/v1/devices/route.js:3:10)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async l (/.next/server/app/api/v1/devices/route.js:1:755)
    at async /node_modules/.pnpm/next@14.0.4_@babel+core@7.23.6_react-dom@18.2.0_react@18.2.0/node_modules/next/dist/compiled/next-server/app-route.runtime.prod.js:6:41960
```

Brandawg93, 2023-12-29:
Hmm. Ok. I've pushed a new test and this time I put lots of console.logs in it. I also disabled the validation so it shouldn't throw that error. We must be getting close!

Brandawg93, 2023-12-29:
I was finally able to add dummy ups devices to my personal server. I found the issue and fixed it. Latest test works!

dimatx, 2023-12-30:
Confirmed fixed!

closed 2023-12-30 by dimatx