Brandawg93/PeaNUT #13: [Feature Request]: Add support for specifying url base

augustozanellato, 2023-12-27:
### Enhancement of issue

I'm using Caddy as a reverse proxy in front of PeaNUT, my intention was to make it serve PeaNUT at path `my.host.tld/nut/` with the following config:
```
route /nut/* {
	uri strip_prefix /nut
	reverse_proxy peanut:8080
}
```
But that doesn't work because nextjs has an hardcoded base path of `/`, so PeaNUT gets stuck at the loading screen

### Solution

Either a way to specify a base path via an environment variable or have PeaNUT use only relative paths. The second option would probably be cleaner but I'm not sure if nextjs supports that, for example using react you'd only need to set `URL_BASE=.`

### Additional Context

Congrats for the great work, PeaNUT looks awesome, I've been searching for a similar tool for quite some time :)

Brandawg93, 2023-12-27:
Good idea! I've created a test version using the tag `:test`. You can try it via `brandawg93/peanut:test`. Try adding a `BASE_PATH` variable and see if that works for you.

augustozanellato, 2023-12-28:
Doesn't seem to work unfortunately. If you push your changed I might work on getting it working next week :)

Brandawg93, 2023-12-28:
Ok. I've pushed it to the main branch. Take a look at aafa3b5ec6b78023181d2de21c458d1ee8bca0e9.

It worked for me when I went to the specified path, but I didn't try with a proxy.

Brandawg93, 2024-01-28:
Feel free to reopen this if you get it working!

closed 2024-01-28 by Brandawg93