Brandawg93/PeaNUT #14: [Feature Request]: Support for ups user

jeffreyswiggins, 2023-12-30:
### Enhancement of issue

No where in the container config can I find config to specify the "ups user".  Most NUT Servers I have run into user "MonUser" by Default and my current setup uses that with a password as well.  I have tried several times to reconfigure with out one in the hopes of making this container connect but the NUT server will not start without one.

### Solution

Support having a UPS User

### Alternatives

Not sure what alternative I have but to not use the container

Brandawg93, 2023-12-31:
You can add `USERNAME` and `PASSWORD` as environment variables to run as a specific user.

https://github.com/Brandawg93/PeaNUT/blob/main/src/app/actions.ts#L10

jeffreyswiggins, 2023-12-31:
The NUT server requires a set up of something like [monuser] which means you have a client connection of:

***@***.*** 1 monuser %password% slave

I have tried starting the NUT server numerous ways with no password (or even monuser) and it just fails to start citing an error in the log need a user or missing parameter.

This [monuser] is not a local user on my Linux server to grant access via the UID GID

Brandawg93, 2023-12-31:
It's my understanding that you don't need to log into the server to read values, only to write. Check out #8 since they had a similar issue.

For my server, I am able to directly telnet in via `telnet localhost 3493` without authentication.

jeffreyswiggins, 2023-12-31:
Okay thank you, after reading through this I will keep looking at it as mine is a Synology on the DSM 7.2 as well so my guess is I am just not getting “lucky”…

Time to play around with it even more…

Brandawg93, 2024-01-21:
If you find any info about this feel free to reopen the issue.

closed 2024-01-21 by Brandawg93