Brandawg93/PeaNUT #17: [Bug]: Portainer (Docker) Health Check Failing

xxirs, 2024-01-07:
### Describe the bug

Portainer (Docker) Health Check Failing

PeaNUT runs fine just health check failing

### Debug Output

ATTACHED PICTURE
[image]

Docker Network
[image]

### Steps to reproduce

1. Run PeaNUT in docker under a network

### Device Type

Asustor 6706T

### UPS Device

CP1500PFCLCD

### Last Working Version

Works, Health Check Fails

Brandawg93, 2024-01-07:
Would you mind sharing your docker command or docker compose yml? That may make debugging easier.

xxirs, 2024-01-07:
> Would you mind sharing your docker command or docker compose yml? That may make debugging easier.

Sure! here you go!

version: "3"

services:
  PeaNUT:
    image: brandawg93/peanut:latest
    container_name: PeaNUT
    restart: unless-stopped
    ports:
      - 8080:8080
    environment:
      - NUT_HOST=192.168.1.5
      - NUT_PORT=3493
      - WEB_PORT=8080
networks:
  default:
    name: lan
    external: true

xxirs, 2024-01-08:
Did some additional testing seems like docker health checks were added in 2.2.0.

2.1.0 does not have this issue and all is good for me, if that helps you any.

Brandawg93, 2024-01-08:
If you docker exec into the peanut container, are you able to send a request to http://localhost:8080/api/ping? That's essentially what the health check is doing.

xxirs, 2024-01-08:
> http://localhost:8080/api/ping

what type of request? docker exec peanut ping  http://localhost:8080/api/ping          ?

xxirs, 2024-01-08:
[image]

From inside the container

xxirs, 2024-01-08:
> http://localhost:8080/api/ping

However, my local IP does work.
[image]

Brandawg93, 2024-01-08:
Interesting. I straight up copied the health check from [Homepage](https://gethomepage.dev/latest/). I'm curious if that works for you. It's also an awesome project. 😀

xxirs, 2024-01-08:
> Interesting. I straight up copied the health check from [Homepage](https://gethomepage.dev/latest/). I'm curious if that works for you. It's also an awesome project. 😀

I use homepage! and the healthcheck does indeed work
[image]

xxirs, 2024-01-08:
> What if you customized the health check in your docker compose yml?
> 
> ```
> healthcheck:
>    test: wget --no-verbose --tries=1 --spider http://localhost:8080/api/ping || exit 1
>    interval: 1m
>    timeout: 3s
>    retries: 3
>    start_period: 20s
> ```
> 
> You could try changing localhost to the ip you used before.

There's parse errors it doesn't seem to like the health checks. My question is why does homepages work? 
[image]

xxirs, 2024-01-08:
> healthcheck:
>    test: wget --no-verbose --tries=1 --spider http://localhost:8080/api/ping || exit 1
>    interval: 1m
>    timeout: 3s
>    retries: 3
>    start_period: 20s

Fixed the parse errors seems to work, main question now is why does homepages work out of the box? Everything is the same with a homepage setup except the LAN IP.
[image]

Brandawg93, 2024-01-08:
I just checked and mine has the same 172 type ip.

My theory is that the healthcheck is checking too quickly and failing. I created a test version of the app that you can install via `brandawg93/peanut:test`. I added extra verbosity to the healthcheck so you can see what is happening. Let me know if that helps!

xxirs, 2024-01-08:
[image]
 Tried the test version same thing, nothing additional in the server log either.

Brandawg93, 2024-01-08:
I just thought of something. Do you have wget installed OUTSIDE the container? In your host environment?

xxirs, 2024-01-08:
I'm actually unsure, and not sure I can check until I get home from work today. PeaNUT is running on my NAS (https://www.asustor.com/en/product?p_id=78).

Brandawg93, 2024-01-08:
Actually, thinking about it, it shouldn't matter if you got the health check working in your compose file. Just thinking out loud 😅. I'll try backing the interval back to 1 minute like your compose. Then it will be exactly the same as what you have working.

Thanks again for helping with this! I'll have a test ready soon but no rush.

xxirs, 2024-01-08:
> I'm actually unsure, and not sure I can check until I get home from work today. PeaNUT is running on my NAS (https://www.asustor.com/en/product?p_id=78).

Looks like it is installed
admin@NAS:/volume1/home/admin $ wget
wget: missing URL
Usage: wget [OPTION]... [URL]...

Try `wget --help' for more options.
admin@NAS:/volume1/home/admin $

Brandawg93, 2024-01-08:
Ok. I'm pretty sure I fixed it!

Context:

Homepage uses the older NextJS pages implementation while peanut uses the newer app router implementation. For some reason, the newer one binds to the external docker ip instead of localhost like you noticed. I simply changed `localhost` to `host.docker.interal` and it works perfectly for me now!

Let me know if it works for you and I will push an official fix.

xxirs, 2024-01-08:
Unfortunately, the fix doesn't work for me.

xxirs, 2024-01-08:
[image]

Brandawg93, 2024-01-08:
Ok. This one has to be it. 😅

NextJS has a HOSTNAME environment variable that has the weird local address. 

Let me know if it works now.

xxirs, 2024-01-08:
> Ok. This one has to be it. 😅
> 
> NextJS has a HOSTNAME environment variable that has the weird local address.
> 
> Let me know if it works now.

Works! 
[image]

Thanks for your help resolving and great project! There isnt any other project super simple just plug and play UPS monitoring.

Brandawg93, 2024-01-08:
Awesome! Thanks again for troubleshooting. v2.2.1 will be live soon!

closed 2024-01-08 by Brandawg93