Brandawg93/PeaNUT #23: [Bug]:  Peanut v2.3.0 Homepage Main

xxirs, 2024-01-20:
### Describe the bug

  Using latest homepage main, peanut was working fine with homepage until the v2.3.0 update. Now it gets these errors inside homepage.

[image]

### Debug Output

[image]

### Steps to reproduce

1. Use latest homepage 
2. + latest peanut

### Device Type

Ubuntu Server Docker

### UPS Device

Cyberpower

Brandawg93, 2024-01-20:
Correct. There is currently an issue with peanut and homepage. It will be fixed in the next version of homepage. V0.8.5.

xxirs, 2024-01-21:
> Correct. There is currently an issue with peanut and homepage. It will be fixed in the next version of homepage. V0.8.5.

homepage just updated the this bug is still present.
Homepage - 0.8.5
Peanut - 2.3.0

To note peanut 2.2.1 works with homepage, 2.3.0 does not.

Brandawg93, 2024-01-21:
Yep. Definitely a bug. Should be fixed now in v2.3.1. Thanks for the heads up!

divemasterjm, 2024-01-21:
Application error: a client-side exception has occurred (see the browser console for more information). in 2.3.1, api works but webpage or api with homepage doesnt work

Brandawg93, 2024-01-21:
Fixed! My bad. 😅 v2.3.2 is up.

divemasterjm, 2024-01-21:
> Fixed! My bad. 😅 v2.3.2 is up.

it works, thanks

brandmanNH, 2024-01-21:
Solved here as well, thank you!

divemasterjm, 2024-01-21:
still getting 
Application error: a client-side exception has occurred (see the browser console for more information). 
on web page load

Brandawg93, 2024-01-21:
@divemasterjm so it’s not working anymore? Could it be a cache issue if it worked just a bit ago?

divemasterjm, 2024-01-21:
> @divemasterjm so it’s not working anymore? Could it be a cache issue if it worked just a bit ago?

i've tried with another explorer, same problem, widget works but no web page

ffeingol, 2024-01-21:
I'm getting similar:

`index-2512dda052eaba9d.js:1 Error: Minified React error #31; visit https://reactjs.org/docs/error-decoder.html?invariant=31&args[]=object%20with%20keys%20%7Bvalue%7D for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at qa (framework-eda4609282c049ac.js:1:56518)
    at e (framework-eda4609282c049ac.js:1:61531)
    at wu (framework-eda4609282c049ac.js:1:73225)
    at xi (framework-eda4609282c049ac.js:1:122263)
    at bs (framework-eda4609282c049ac.js:1:109372)
    at gs (framework-eda4609282c049ac.js:1:109300)
    at vs (framework-eda4609282c049ac.js:1:109163)
    at as (framework-eda4609282c049ac.js:1:105943)
    at ls (framework-eda4609282c049ac.js:1:104494)
    at S (framework-eda4609282c049ac.js:1:138467) `

This is with the default widget or a custom one using the API.  I'm on HomePage V0.8.6 and PeaNUT V2.3.2.

Brandawg93, 2024-01-21:
@divemasterjm I've created a new test version that may fix your issue. You can try it by using `brandawg93/peanut:test`. Just make sure to pull the image before restarting the container.

@ffeingol Is that on homepage or peanut? The homepage issue should be fixed. Try checking the peanut api to see if `api/v1/devices` has lots of `value: {}` keys in it.

divemasterjm, 2024-01-21:
> @divemasterjm I've created a new test version that may fix your issue. You can try it by using `brandawg93/peanut:test`. Just make sure to pull the image before restarting the container.
>
> @ffeingol [...]

it works now

ffeingol, 2024-01-21:
Hello,

Sorry, that's on homepage.  In my custom URL if I use /api/v1/devices/ I get nill for all the values.  My current config uses /api/v1/devices/ups  If I browse to either URL I see valid json values.  The only different between the URL's is if I leave the /ups off then the json values are surrounded by [] which homepage does not like at all (get the nil)

Brandawg93, 2024-01-21:
@ffeingol that is expected since you can only have one device per widget. If you want to use more than one device, just make more than one service and specify the device names in each widget endpoint.

Brandawg93, 2024-01-21:
@divemasterjm Thanks for testing! v2.3.3 is up now.

Brandawg93, 2024-01-21:
@ffeingol Can you post a sample of your api output?

closed 2024-01-21 by Brandawg93

divemasterjm, 2024-01-21:
> @divemasterjm Thanks for testing! v2.3.3 is up now.

Thanks, it works now

ffeingol, 2024-01-21:
Sure, here you go.  This is the full output.

{"battery.charge":{"value":"100"},"battery.charge.low":{"value":"10"},"battery.charge.warning":{"value":"50"},"battery.date":{"value":"2001/09/25"},"battery.mfr.date":{"value":"2011/04/02"},"battery.runtime":{"value":"12824"},"battery.runtime.low":{"value":"120"},"battery.type":{"value":"PbAc"},"battery.voltage":{"value":"27.0"},"battery.voltage.nominal":{"value":"24.0"},"device.mfr":{"value":"American Power Conversion"},"device.model":{"value":"Back-UPS BX1000G"},"device.serial":{"value":"3B1114X48132"},"device.type":{"value":"ups"},"driver.name":{"value":"usbhid-ups"},"driver.parameter.pollfreq":{"value":"30"},"driver.parameter.pollinterval":{"value":"2"},"driver.parameter.port":{"value":"auto"},"driver.parameter.synchronous":{"value":"auto"},"driver.version":{"value":"2.8.0"},"driver.version.data":{"value":"APC HID 0.98"},"driver.version.internal":{"value":"0.47"},"driver.version.usb":{"value":"libusb-1.0.26 (API: 0x1000109)"},"input.sensitivity":{"value":"medium"},"input.transfer.high":{"value":"139"},"input.transfer.low":{"value":"88"},"input.voltage":{"value":"122.0"},"input.voltage.nominal":{"value":"120"},"ups.beeper.status":{"value":"disabled"},"ups.delay.shutdown":{"value":"20"},"ups.firmware":{"value":"869.L2 .D"},"ups.firmware.aux":{"value":"L2"},"ups.load":{"value":"3"},"ups.mfr":{"value":"American Power Conversion"},"ups.mfr.date":{"value":"2011/04/02"},"ups.model":{"value":"Back-UPS BX1000G"},"ups.productid":{"value":"0002"},"ups.realpower.nominal":{"value":"600"},"ups.serial":{"value":"3B1114X48132"},"ups.status":{"value":"OL"},"ups.test.result":{"value":"No test initiated"},"ups.timer.reboot":{"value":"0"},"ups.timer.shutdown":{"value":"-1"},"ups.vendorid":{"value":"051d"}}

Let me know if that helps.

Brandawg93, 2024-01-21:
Ah. Looks like you aren’t on the latest version. You may need to pull the image again from docker.

ffeingol, 2024-01-21:
Sorry about that.  Thought I had pulled it.  All good now and thanks for the effort.