Brandawg93/PeaNUT #24: [Feature Request]: Custom Fields in HomePage

brandmanNH, 2024-01-21:
### Enhancement of issue

As per Homepage Version 0.8.6 (with PeaNUT Version 2.3.2) the only allowed fields are: ["battery_charge", "ups_load", "ups_status"] (https://gethomepage.dev/latest/widgets/services/peanut/)

It is possible to remove one or more of the allowed fields in the widget configuration (https://github.com/gethomepage/homepage/issues/2697), but you cannot add any other field (like battery runtime or something else) out of the key list in PeaNUT.

### Solution

Possibility to add max. 4 custom fields in the PeaNUT widget (like described in #2697 listed above)

### Alternatives

-

### Additional Context

-

Brandawg93, 2024-01-22:
For this, I would suggest using the `customapi` widget and creating your own. Many of the fields in the api response are specific to your device, so others may not have them.

Here is an example with battery runtime:

```
        widget:
            type: customapi
            url: http://{localhost}:{port}/api/v1/devices/ups
            mappings:
            - field: battery.charge
              label: Battery Charge
              format: percent
            - field: battery.runtime
              label: Battery Runtime
              format: text
            - field: ups.load
              label: UPS Load
              format: percent
            - field: ups.status
              label: UPS Status
              format: text
              remap:
                - value: OL
                  to: Online
                - value: OB
                  to: On Battery
                - value: LB
                  to: Low Battery
                - any: true
                  to: Unknown
```

brandmanNH, 2024-01-23:
Some additional manual configuration in JS is needed to get the value "battery.runtime" which has to be scaled with 0.0166667 to get the runtime from seconds to minutes.

By now there is no rounding function available in the customapi widget in homepage.

Thank you @Brandawg93 :)

closed 2024-01-23 by brandmanNH

brandmanNH, 2024-01-23:
Found something a lot easier:

        - field: battery.runtime
          label: Runtime
          format: float
          scale: 1/60
          suffix: "Min"

Works as expected an no additional manual configuration in JS needed.
Mentioned by @ffeingol in https://github.com/Brandawg93/PeaNUT/issues/23