Brandawg93/PeaNUT #26: [Bug]: `2.3.3` healthcheck failing when binding to IPv6

troykelly, 2024-01-22:
### Describe the bug

The docker healthcheck is failing in the most recent docker release. I think this is because the service is being bound to "::".

ie:

```yaml
  nut_dashboard:
    <<: *default-opts
    image: brandawg93/peanut:latest
    restart: unless-stopped
    network_mode: host
    environment:
      - "TZ=Australia/Sydney"
      - NUT_HOST=localhost
      - NUT_PORT=3493
      - WEB_PORT=3480
      - PORT=3480
      - "HOSTNAME=::"
```

A working healthcheck is:

```yaml
    healthcheck:
      test:
        [
          "CMD-SHELL",
          'if echo "$${HOSTNAME}" | grep -q '':''; then WGET_HOST="http://[$${HOSTNAME}]:$${PORT}"; else WGET_HOST="http://$${HOSTNAME}:$${PORT}"; fi; wget --spider --quiet --tries=1 --timeout=30 $$WGET_HOST || exit 1',
        ]
      interval: 1m
      timeout: 10s
      retries: 3
      start_period: 40s
```

### Debug Output

```text
nutupsd-nut_dashboard-1  |    ▲ Next.js 14.0.4
nutupsd-nut_dashboard-1  |    - Local:        http://[::1]:3480
nutupsd-nut_dashboard-1  |    - Network:      http://[::]:3480
nutupsd-nut_dashboard-1  | 
nutupsd-nut_dashboard-1  |  ✓ Ready in 295ms
```

### Steps to reproduce

1. Define `HOSTNAME` as "::"
2. Start service
3. Healthcheck fails

### Device Type

N/A

### UPS Device

N/A

Brandawg93, 2024-01-22:
It looks like you are defining HOSTNAME in your compose as "::". Unfortunately, HOSTNAME is a reserved environment variable in the newest version of nextjs. What happens if you remove that line in your compose?

troykelly, 2024-01-22:
It goes back to not listening on all interfaces.

```text
nutupsd-nut_dashboard-1  |    ▲ Next.js 14.0.4
nutupsd-nut_dashboard-1  |    - Local:        http://dns002:3480
nutupsd-nut_dashboard-1  |    - Network:      http://127.0.1.1:3480
nutupsd-nut_dashboard-1  | 
nutupsd-nut_dashboard-1  |  ✓ Ready in 322ms
```

```yaml
      - "HOSTNAME=::"
```

Is working fine - the service is great (thank you!). It's just the health check that needs to handle the IPv6 address. The health check I'm using (in the original issue comment) is working fine as well.

Brandawg93, 2024-01-22:
I tested the above health check, but it didn't work for my particular setup. Ultimately, the provided health check is just a template to cover most users. And it seems like you've found the one that works for you! I would suggest to just continue using that with your particular setup.

closed 2024-01-25 by Brandawg93