Brandawg93/PeaNUT #28: [Bug]:

sahara101, 2024-01-25:
### Describe the bug

not working after update. page does not load, but no errors in the log
compose is this:

version: '3'
services:
  nut_dashboard:
    image: brandawg93/peanut:latest
    container_name: PeaNUT
    restart: unless-stopped
    ports:
      - 6543:8080
    environment:
      - NUT_HOST=192.168.1.150
      - NUT_PORT=3493
      - WEB_PORT=6543

Thanks!

### Debug Output

> peanut@2.3.4 start
> PORT=$WEB_PORT node server.js
   ▲ Next.js 14.1.0
   - Local:        http://3f6de5575163:6543
   - Network:      http://172.29.0.2:6543
 ✓ Ready in 511ms

### Steps to reproduce

1. Watchtower update

### Device Type

Synology docker - portainer

### UPS Device

USB

Brandawg93, 2024-01-25:
Looks like your port configuration is incorrect. It should be `6543:6543`.

sahara101, 2024-01-25:
hmm.. I have not changed since first creating it. Working again, thanks!

closed 2024-01-25 by Brandawg93