Brandawg93/PeaNUT #30: [Bug]: 2.3.4 now working

vobi1024, 2024-01-26:
### Describe the bug

Latest version not working, only the gray background with the spinning wheel loads.

### Debug Output

2024/01/26 14:23:05 | stdout | [32m[1m✓[22m[39m Ready in 171ms
-- | -- | --
2024/01/26 14:23:04 | stdout |  
2024/01/26 14:23:04 | stdout | - Network:      http://172.17.0.7:8080
2024/01/26 14:23:04 | stdout | - Local:        http://brandawg93-peanut-1:8080
2024/01/26 14:23:04 | stdout | [1m[38;2;173;127;168m ▲ Next.js 14.1.0[39m[22m

### Steps to reproduce

1. Start container
2. try to open webpage

### Device Type

Synology Docker

### UPS Device

Synology UPS server, LAN

### Last Working Version

2.3.3

Brandawg93, 2024-01-26:
Do you see any error in the browser console? Also, does the latest test version work for you? `brandawg93/peanut:test`

vobi1024, 2024-01-26:
> Do you see any error in the browser console? Also, does the latest test version work for you? `brandawg93/peanut:test`

No, there is only a timeout. But yes, the test version works!

Brandawg93, 2024-01-28:
Great! v2.4.0 is now live with a few extra features as well.

closed 2024-01-28 by Brandawg93