Brandawg93/PeaNUT #31: [Feature Request]: Identify UPS by Name instead of Model

keezppc, 2024-01-27:
### Enhancement of issue

I have 3 of the exact same UPS Model, labeled in NUT Server as Top, Middle, Bottom.  PeaNUT not show this information when selecting a device, only the UPS model.  Would it be possible to have the dropdown show the name of the UPS as pulled from NUT server instead of the Model Name?
[image]

### Solution

Option to show the name of the UPS from NUT server instead of the model name

Brandawg93, 2024-01-28:
Good idea! I've switched it to use the description (desc) of each ups in your `ups.conf`. That way someone could leave out the description and have the app fall back to the previous implementation. It's currently live on a test version that you can try via `brandawg93/peanut:test`. Let me know what you think!

keezppc, 2024-01-28:
Looks great! Thanks for the quick implementation

Brandawg93, 2024-01-28:
Great! This is now live in v2.4.0.

closed 2024-01-28 by Brandawg93