Brandawg93/PeaNUT #35: [Bug]: Blank webpage when compiling from source

remz1337, 2024-02-05:
### Describe the bug

The server starts without error and accessing `http://localip:3000` shows a blank page with a small black circle in the top left. Web logs are showing that a bunch of script (in `http://localip:3000/_next/static/`) could not be loaded.

I have nut-client installed and working locally (server runs on another machine), and returning the UPS data on `upsc` command

### Debug Output

```
GET
http://192.168.1.123:3000/_next/static/media/c9a5bc6a7c948fb0-s.p.woff2
[HTTP/1.1 404 Not Found 168ms]

```

### Steps to reproduce

I followed the steps to build from source on a fresh Ubuntu 22.04 LXC:

as `root`:
```
apt install git
curl -fsSL https://deb.nodesource.com/setup_21.x | bash - && apt-get install -y nodejs
git clone https://github.com/Brandawg93/PeaNUT.git
cd PeaNUT
npm i -g pnpm
pnpm i
pnpm run build
node .next/standalone/server.js
```

### Device Type

Ubuntu 22.04 LXC

### UPS Device

CyberPower over USB

remz1337, 2024-02-12:
@Brandawg93 Do I need to install nut manually? And will it work if nut is configured as netclient only (I already have a nut server running on another machine)

Brandawg93, 2024-02-13:
Yes, you will need to install nut manually. And I do believe it has to be in `server` mode to work properly. If you change it, and it works let me know and I'll update the docs.

vhsdream, 2024-02-17:
Chiming in here that I also wanted to use PeaNUT to display the stats for an already-configured NUT remote server. I've only figured out how to query the device via the api by appending NUT_HOST=<ip of remote NUT host> before the `node .next/standalone/server.js` command. However when navigating to the main page it is still blank with the black circle as OP describes.

vhsdream, 2024-06-06:
I submitted a PR (#55) that fixes this issue. Not a dev, just someone who spent too much time googling what was happening. 😄

Brandawg93, 2024-06-06:
Cool! If someone else can confirm this fixes the issue for them, I will merge it into the main branch. I do think I may need to rework the `Dockerfile` to accommodate this, so just want to make sure this fix works for others.

vhsdream, 2024-06-06:
Actually I think you don't need to change the Dockerfile (not sure as I don't use Docker) because the copy commands are already [there](https://github.com/Brandawg93/PeaNUT/blob/3812fe680b561941d7ee69fbc24411aa2c8b28a9/Dockerfile#L33-L34).

The fact that the files are not copied when server.js is run is mentioned [here](https://nextjs.org/docs/pages/api-reference/next-config-js/output#automatically-copying-traced-files), in a section where they talk about files being automatically copied. I find that kind of misleading even if they do mention it.

Oh and I also tested PeaNUT with a NUT server running on a different server, and as long as I specify the NUT_HOST it will work, so @remz1337 it will work as a client-only setup too.

Brandawg93, 2024-06-06:
I read that as well regarding copying. Ironically, the static folder is automatically copied on my Mac but not on the docker builds of Ubuntu. Not sure why it’s different for Linux. Either way, I always like to get a second outside confirmation just in case.

Brandawg93, 2024-06-06:
I went ahead and merged as I don't think this will need a new release for those compiling from source. If anyone still has issues, feel free to reopen.

closed 2024-06-06 by Brandawg93

remz1337, 2024-06-13:
tested successfully. Thank you

commit 9daaaab5 (2024-06-06): Merge pull request #55 from vhsdream/main

add copy command to package.json - fixes #35