Brandawg93/PeaNUT #40: [Bug]: Docker logs not capturing http errors

jbestt, 2024-02-22:
### Describe the bug

Peanut is returning Internal Server Errors, but I can't perform root cause analysis due to the lack of logs.

upsc returns good telemetry for all of the configured ups's.

### Debug Output

```
root@nut:~/peanut# docker logs peanut

> peanut@2.5.0 start
> PORT=$WEB_PORT node server.js

   ▲ Next.js 14.1.0
   - Local:        http://e40beb5a33c6:8080
   - Network:      http://172.18.0.2:8080

 ✓ Ready in 303ms
```

### Steps to reproduce

1. Configure nut with three UPS devices via snmp-ups
2. Install PeaNUT via Docker
3. Visit PeaNUT page

### Device Type

Debian LXC Container

### UPS Device

Various APC via snmp-ups

### Last Working Version

n/a

nicrame, 2024-05-18:
I can confirm situation. I tried to check for logs by connecting inside container, but /var/log/ is empty.

Brandawg93, 2024-07-03:
Do you see more logs if you clone the repo and run `npm run dev`?

zzz676, 2024-07-16:
Same thing, won't start. and a circle just spins at the address in the browser

dasilvatek, 2024-10-22:
@jbestt @nicrame are you guys using 8080 for any other running containers? That could be the issue. I'm already running another application on 8080 so I had to adapt like this and hit the browser on port 8888:

`docker run -p 8888:8080 -e NUT_HOST=localhost -e NUT_PORT=3493 -e WEB_HOST=10.0.0.10 -e WEB_PORT=8080`

```
> peanut@3.0.0 start
> HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js

  ▲ Next.js 14.2.13
  - Local:        http://localhost:8080
  - Network:      http://0.0.0.0:8080

 ✓ Starting...
 ✓ Ready in 91ms
```

I only joined the party at 2.6 and confirm no issues with it or the latest 3.0 if you want to try it out. I'm here to help but you'll have to paste in more details to troubleshoot.

Brandawg93, 2024-10-28:
This issue has been fixed in the latest version (v3.2.1). Let me know if you still have issues.

closed 2024-10-28 by Brandawg93