Brandawg93/PeaNUT #48: [Bug]: Endless spinner on blank page with no errors

MaxLaumeister, 2024-04-10:
### Describe the bug

Peanut does not do anything when I connect to it - there is only a spinner in the center of the page.

(Thank you for your time and effort in maintaining this plugin!)

### Debug Output

When loading the PeaNUT Web UI, Chrome's JavaScript console is blank. Chrome's network log shows that all resources loaded properly with `200` status.

**Docker log:**

```
PeaNUT  | > peanut@2.5.0 start
PeaNUT  | > PORT=$WEB_PORT node server.js
PeaNUT  | 
PeaNUT  |    ▲ Next.js 14.1.0
PeaNUT  |    - Local:        http://a51845303b51:8080
PeaNUT  |    - Network:      http://172.23.0.2:8080
PeaNUT  | 
PeaNUT  |  ✓ Ready in 47ms
```

### Steps to reproduce

1. Verify that NUT is properly serving on port 3493.
2. Use this `docker-compose.yml`:
    ```
    version: '3'
    services:
      peanut:
        image: brandawg93/peanut:latest
        container_name: PeaNUT
        restart: unless-stopped
        ports:
          - 8080:8080
        environment:
          - NUT_HOST=localhost
          - NUT_PORT=3493
          - WEB_PORT=8080
    ```
3. Run `docker compose up -d`
4. Visit `localhost:8080` via SSH tunnel
5. Endless spinner

### Device Type

Intel NUC

### UPS Device

Eaton 9SX2000; usbhid-ups

Brandawg93, 2024-04-20:
Does the api work for you? As in `{baseUrl}/api/v1/devices`

MaxLaumeister, 2024-04-22:
When I visit `/api/v1/devices`, it gives me an HTTP error 500.

Brandawg93, 2024-04-22:
Ok. And you don’t see any additional error message? My hypothesis is that it can’t connect to your NUT server. Are you able to connect to the NUT server directly?

MaxLaumeister, 2024-04-22:
Correct, there is no additional error message.

I am able to connect to the NUT server directly. When I run `upsc eaton@localhost` from the same machine, and when I `upsc eaton@[nut server ip]` from another machine on the same LAN, it properly returns the UPS stats. 

Please note that I just changed the port to `8090` in my docker compose file, because I started running a different service on `8080` in the mean time. This is the new docker compose file:

```
version: '3'
services:
  peanut:
    image: brandawg93/peanut:latest
    container_name: PeaNUT
    restart: unless-stopped
    ports:
      - 8090:8090
    environment:
      - NUT_HOST=localhost
      - NUT_PORT=3493
      - WEB_PORT=8090
```

And here are the screenshots of what happens when I connect via browser:

[image]

[image]

Brandawg93, 2024-04-22:
Out of curiosity, what happens if you keep `WEB_PORT` as the default 8080 and use `8090:8080` for the docker ports? Could be a issue in the Dockerfile on my end 😄

MaxLaumeister, 2024-04-22:
No prob! I updated my dockerfile:

```
version: '3'
services:
  peanut:
    image: brandawg93/peanut:latest
    container_name: PeaNUT
    restart: unless-stopped
    ports:
      - 8090:8080
    environment:
      - NUT_HOST=localhost
      - NUT_PORT=3493
      - WEB_PORT=8080
```

Then I did a `docker compose down` and `docker compose up -d`. The same behavior is exhibited when I load `/` and `/api/v1/devices` in the browser.

I also opened up port `8090` temporarily and connected directly over LAN with my web browser, to rule out the SSH tunnel causing any issues. But I still get the exact same behavior in the browser.

Brandawg93, 2024-04-22:
Hmm. What happens if you change `NUT_HOST` to the ip of the host device (instead of localhost)? Maybe docker isn’t properly translating localhost.

MaxLaumeister, 2024-04-22:
Oh! That did the trick. I changed `NUT_HOST` from `localhost` to the ip of the ethernet adapter of the machine hosting docker hosting peanut. Now both the dashboard and the API are working.

Thank you for helping me diagnose this!

closed 2024-04-22 by MaxLaumeister