Brandawg93/PeaNUT #50: [Bug]: What is my device name for the homepage widget?

Kirk1984, 2024-05-12:
### Describe the bug

First of all, Everything is working fine via the WebUI (See Screenshot). I want to integrate the Data into my (get)homepage, but i cannot get my Device name. The UPS is connected to my QNAP NAS.

The default name `ups` isn't working 

[image]

### Debug Output

The api output:

[image]

### Steps to reproduce

1. Have an ups by eaton connected to a qnap nas? I don't know :)

### Device Type

QNAP

### UPS Device

Eaton 3S 850

### Last Working Version

All

Brandawg93, 2024-05-17:
I'm not sure on QNAP devices. If they are similar to Synology, then they run their own custom nut-upsd that may have different configuration values from standard setups.

Kirk1984, 2024-05-17:
Thanks! That was the hint i needed. :)

I thought the ups name was something that PeaNUT manages. QNAP calls the ups `qnapups` by default. With that it works.

Brandawg93, 2024-05-17:
Glad to hear it's working for you!

closed 2024-05-17 by Brandawg93