Brandawg93/PeaNUT #52: [Bug]: Device Manufacturer/ Model/ Serial/ Runtime is missing

iskandarluqman, 2024-05-15:
### Describe the bug

Hello. Before that I would like to say everything besides that is showing perfectly
[image]

Just only some of it are missing and not showing the name is kinda bothering me. I am migrating from webNUT to peaNUT. 

[image]

Webnut shows the model no issue. So far if I recalled correctly, there isnt any configuration file for webnut so the name must be from the device itself. Just I dont know why peaNUT is not showing the name. 

### Debug Output

Missing Device Manufacturer/ Model/ Serial/ Runtime

### Steps to reproduce

1. Just create a new docker container with this add-on of configuration
      - WEB_PORT=443
      - USERNAME=admin
      - PASSWORD=admin
      - BASE_PATH=test.example.com

networks:
  default:
    name: custom1
    external: true

### Device Type

Desktop

### UPS Device

offline / line interactive, nutdrv_qx

Brandawg93, 2024-05-15:
That's interesting. What do you see when using the API?

iskandarluqman, 2024-05-16:
How do I test the API actually? Sorry I am not familiar. I tried by clicking the API docs
[image]

I am not sure what is the device ID.

Brandawg93, 2024-05-16:
Hm. I've never seen a ups that didn't report its model. I've published a new test version that should use your device description in the event that there is no model. Could you try it out? `brandawg93/peanut:test`

iskandarluqman, 2024-05-16:
[image]

That definitely did the trick. Thank you so much! However I did messed around with changing the driver for my UPS, as I lookup for my UPS, nutdrv_qx was the one to be used as my UPS comes with Viewpower software. Even nut scanner recommended this. Then I decided to try random drivers and I end up with blazer_usb driver. Somewhat this driver is better as it show device.model etc. 
[image]

Not sure why but hey it works in the end. Just though letting you know. Thank you again!

Brandawg93, 2024-05-16:
Even better! Thanks for testing my changes, though. I'll close this as completed since your problem is resolved but will still publish a new official version for anyone else with this issue.