Brandawg93/PeaNUT #57: [Bug]: BASE_PATH Doesn't Seem To Work

Majoraslayer, 2024-06-18:
### Describe the bug

I'm trying to set the base path variable per the instructions for my reverse proxy.  If properly functioning, I should be able to access it locally at 192.168.1.69:8008/pnut.  However, no matter how I set this variable I seem to get a 404 error served out by PeaNUT itself when navigating via the base path.  The guide doesn't have any actual examples on how to use it.  I've tried all of the following formatting in my Docker run command:

```
-e BASE_PATH=/pnut
-e BASE_PATH=/pnut/
-e BASE_PATH="/pnut"
-e BASE_PATH="/pnut/"
-e BASE_PATH=pnut
-e BASE_PATH="pnut"
```

Is the feature broken, or are the instructions to use it just missing important context?

### Debug Output

N/A

### Steps to reproduce

1. Use the docker run command and set the BASE_PATH environment variable to anything.

### Device Type

Debian x86

### UPS Device

TrippLite Smart1300CLDt

Brandawg93, 2024-07-03:
I've personally never used it myself, but I've heard from others that it's working for them. Not sure if this helps, but here are some resources.

https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://nextjs.org/docs/app/api-reference/next-config-js/basePath&ved=2ahUKEwjvkYD2i4uHAxWxFlkFHUfrAe4QFnoECAgQAQ&usg=AOvVaw3Gav0NLNi7JXw0I050D492

https://github.com/Brandawg93/PeaNUT/blob/main/next.config.js#L9

Brandawg93, 2024-10-22:
Closing as stale.

closed 2024-10-22 by Brandawg93

Brandawg93, 2024-11-25:
That's good to know. So users may need to compile from source for the time being until I can look into it again.

artivis, 2025-03-27:
Just ran into this issue,

> This value must be set at build time and cannot be changed without re-building

This should be made explicit in the readme imho.

Brandawg93, 2025-03-28:
You're right. I haven't updated it because I'm redoing it. 😄 
https://github.com/Brandawg93/PeaNUT/pull/215

Brandawg93, 2025-03-29:
Ok. I finally got around to making a test build. Could someone try it and let me know if it works for you?
`brandawg93/peanut:test`

artivis, 2025-03-29:
I gave this a quick shot and it doesn't seem to work, resulting in `redirected you too many times` (chrome) or `The page isn’t redirecting properly` (firefox).
I can't guarantee that the issue doesn't come from my reverse proxy tho, I'm still learning :)

Brandawg93, 2025-03-29:
> I can access peanut fine without going through the reverse proxy.

Do you mean the `BASE_PATH` works when you don't use the reverse proxy?

This worked in my local testing as well. I haven't tried my actual production copy. I could test with that in a bit.

artivis, 2025-03-29:
No I meant that I can still access peanut fine on `IP:port` with no sub-folder despite the environment variable being set.
Trying `IP:port/peanut` returns a 404.

absinthe4869, 2025-08-22:
I'm not familiar with frontend development, just offers some search results.

It seams like that current implemention uses 'process.env.BASE_PATH', which might not be exposed to client and it is mentioned here:
https://stackoverflow.com/questions/67240908/how-to-set-base-url-based-on-environment-variable-in-next-js

In the mean time, change 'process.env.BASE_PATH' to 'process.env.NEXT_PUBLIC_BASE_PATH' is also no effect, since all 'NEXT_PUBLIC_*' variables get inlined during the build step, which is mentioned here:
https://github.com/vercel/next.js/discussions/16059

The two reference links give 2 methods might be help
- use global.process.env and hack around Next.js' build-time variables
- use package next-runtime-env

Hope this will help you.

absinthe4869, 2025-08-24:
After more tests, I'd like to offer another method to achieve same target.

```
# Assume use base url "/peanut"

# Reverse proxy configuration
location /peanut {
    proxy_pass          http://localhost:8080/peanut;

    proxy_http_version  1.1;

    proxy_set_header    Host               $host;
    proxy_set_header    X-Real-IP          $remote_addr;
    proxy_set_header    X-Forwarded-For    $proxy_add_x_forwarded_for;
    proxy_set_header    X-Forwarded-Proto  $scheme;
}

# Rewrite static assets when and only when the referer comes from PeaNUT
if ($http_referer ~ ^https?://[^/]+/peanut/?$) {
    set $peanut_base_url /peanut;   # NOTE: variable in Nginx is globally visible
    
    rewrite  ^/_next/static/(.+)    $peanut_base_url/_next/static/$1;
    rewrite  ^/favicon.ico([?].+)*  $peanut_base_url/favicon.ico$1;
    rewrite  ^/icon.svg([?].+)*     $peanut_base_url/icon.svg$1;
}
```

The method has the following advantages:
- Rewrite is transparent to client and never modifies the actual files, which maintains consistency with reverse proxy with subdomain.
- It avoids preload issue we discussed earlier and will not generate 404 error at server logs.
- It doesn't requires to set `proxy_set_header Accept-Encoding "";` and can support compress transfer method like gzip.
- Apache, Nginx and Caddy all support rewrite without addons.

There is 2 possible problem with this method:
- According to Nginx wiki(archived at https://github.com/nginxinc/nginx-wiki/blob/master/source/start/topics/depth/ifisevil.rst), use `if` in `location` block might introduce safety risk. If must use, `if` should be put into `server` block(which is what I do here).
- It relies on `$http_referer`, which is not reliable enough actually and can easily changed. In my view this is not a problem since we are in reverse proxy and all header attr should be under control.

I'd like to take your advice about the new configuration method.

Brandawg93, 2025-08-24:
Awesome! This is very helpful. I've updated the wiki with this info. Thanks for working on it! I'm going to merge the work soon since it shouldn't break any current implementations. It'll be available in the next release.