Brandawg93/PeaNUT #65: Docker in background with IP address instead of the name

sysadmin-info, 2024-08-01:
### Enhancement of issue

See the solution

### Solution

sudo docker run -d -p 8080:8080 --restart unless-stopped --env NUT_HOST=10.10.0.109 --env NUT_PORT=3493 --env WEB_PORT=8080 brandawg93/peanut

### Alternatives

No alternative

Brandawg93, 2024-08-04:
I'm a little confused on what the issue is here?

sysadmin-info, 2024-08-04:
No solution. I provided a fix how to run the Docker. You need an IP address where NUT is installed. Not always you have to run the Docker in the same host with NUT. I had no idea where to put this information. It would be good to put this into README.md file. Not everyone have to know that it is possible.

closed 2024-08-04 by Brandawg93