Brandawg93/PeaNUT #66: [Feature Request]:

Scott8586, 2024-08-05:
### Enhancement of issue

Provide a  linux/arm/v7 docker image

### Solution

Provide a linux/arm/v7 docker image

Brandawg93, 2024-08-05:
I've actually tried several times to get arm/v7 working again. For some reason, it doesn't work on the latest version of node and MacOS (my local build machine).

Brandawg93, 2024-10-22:
Ironically, nextjs released version 15 today which adds support for WebAssembly SWC. I tried it and it works just fine. Feel free to try the latest [test version](https://hub.docker.com/layers/brandawg93/peanut/test/images/sha256-ce2c037ea55895ec01dc8b1907b2f344ab8eea89063a9dbef7fd88544158105a?context=repo) with armv7 support.

https://github.com/Brandawg93/PeaNUT/pull/80

Brandawg93, 2024-10-22:
armv7 has been enabled for v3.1.0 and greater.

closed 2024-10-22 by Brandawg93

mookeyj79, 2024-10-22:
Works great! Thank you!