Brandawg93/PeaNUT #74: [Bug]: UPS Serial Number is not fully displaying

dasilvatek, 2024-10-17:
### Describe the bug

The UPS Serial Number is only displaying the first digit via PeaNUT. The Operating System, NUT server and client all resolve the full serial number as evidenced in debug output, but this data is being mangled/ignored by PeaNUT possibly due to alphanumeric values (since it cuts off at the first letter in the serial).

### Debug Output

Check USB device information:
```
~$ lsusb -D /dev/bus/usb/001/003
Device: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x051d American Power Conversion
  idProduct          0x0002 Uninterruptible Power Supply
  bcdDevice            0.90
  iManufacturer           1 American Power Conversion
  iProduct                2 Back-UPS XS 1500G FW:866.L9 .D USB FW:L9 
  iSerial                 3 4B1707P21318  
```

Check NUT server configuration for this UPS device:
```
~$ cat /etc/nut/ups.conf
pollinterval = 1
maxretry = 3

[myups]
    driver = usbhid-ups
    port = auto
    desc = "Back-UPS XS 1500G"
    vendorid = 051d
    productid = 0002
    serial = 4B1707P21318
```

Check NUT client configuration for this UPS device:
```
~$ upsc myups |grep serial
device.serial: 4B1707P21318  
driver.parameter.serial: 4B1707P21318
ups.serial: 4B1707P21318 
```

Run PeaNUT and confirm readiness:
```
~$ docker run -d --name peanut --network monitoring --restart unless-stopped -p 8888:8080 -e TZ=America/New_York -e NUT_HOST=10.0.0.10 -e NUT_PORT=3493 -e WEB_HOST=0.0.0.0 -e WEB_PORT=8080 brandawg93/peanut~

~$ docker logs peanut
> peanut@2.6.1 start
> HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js

  ▲ Next.js 14.2.4
  - Local:        http://localhost:8080
  - Network:      http://0.0.0.0:8080

 ✓ Starting...
 ✓ Ready in 107ms
```

Check PeaNUT device details via API will display only the first digit (ie: 4) for all 3 parameters:
```
~$ curl -X GET http://localhost:8888/api/v1/devices/myups
{"battery.charge":100,"battery.charge.low":"10","battery.charge.warning":50,"battery.date":2001,"battery.mfr.date":"2017/02/18","battery.runtime":15750,"battery.runtime.low":"120","battery.type":"PbAc","battery.voltage":27.2,"battery.voltage.nominal":24,"device.mfr":"American Power Conversion","device.model":"Back-UPS XS 1500G","device.serial":4,"device.type":"ups","driver.debug":"0","driver.flag.allow_killpower":"0","driver.name":"usbhid-ups","driver.parameter.pollfreq":30,"driver.parameter.pollinterval":1,"driver.parameter.port":"auto","driver.parameter.productid":2,"driver.parameter.serial":4,"driver.parameter.synchronous":"auto","driver.parameter.vendorid":51,"driver.state":"updateinfo","driver.version":2.8,"driver.version.data":"APC HID 0.100","driver.version.internal":0.52,"driver.version.usb":"libusb-1.0.27 (API: 0x100010a)","input.sensitivity":"medium","input.transfer.high":"139","input.transfer.low":"88","input.voltage":123,"input.voltage.nominal":120,"ups.beeper.status":"disabled","ups.delay.shutdown":"20","ups.firmware":866,"ups.firmware.aux":"L9","ups.load":"0","ups.mfr":"American Power Conversion","ups.mfr.date":2017,"ups.model":"Back-UPS XS 1500G","ups.productid":2,"ups.realpower.nominal":865,"ups.serial":4,"ups.status":"OL","ups.test.result":"No test initiated","ups.timer.reboot":"0","ups.timer.shutdown":-1,"ups.vendorid":51}
```

Interestingly, checking the PeaNUT API for single parameters will display the accurate serial number:
```
~$ curl -X GET http://localhost:8888/api/v1/devices/myups/var/device.serial
"4B1707P21318"

~$ curl -X GET http://localhost:8888/api/v1/devices/myups/var/driver.parameter.serial
"4B1707P21318"

~$ curl -X GET http://localhost:8888/api/v1/devices/myups/var/ups.serial
"4B1707P21318"
```

Checking PeaNUT via web UI will display only the first digit (ie: 4):
[image]
[image]
[image]
[image]

### Steps to reproduce

1. Install NUT and configure Driver
2. Verify UPS information including Serial Number
3. Install PeaNUT Docker and confirm functionality
4. Confirm UPS Serial Number in PeaNUT

### Device Type

Ubuntu 24.04.1 LTS and kernel 6.8.0-47-generic on Asus PN62S

### UPS Device

APC UPS Back-UPS XS 1500G and usbhid-ups libusb-1.0.27

Brandawg93, 2024-10-17:
Thanks for the report! Could you try my latest version? I may have fixed this already.

https://github.com/Brandawg93/PeaNUT/releases/tag/v3.0.0-beta.0

dasilvatek, 2024-10-17:
@Brandawg93 I had trouble finding settings.yml but hacked my around and got the beta container running.

Sorry to report the same exact results as initially documented -- not showing full serial number via all the same methods  except for API single parameters which are successful.

How can I help track down this glitch? Thanks for your help.
```
wget https://github.com/Brandawg93/PeaNUT/archive/refs/tags/v3.0.0-beta.0.tar.gz
tar xvzf v3.0.0-beta.0.tar.gz
cd PeaNUT-3.0.0-beta.0
docker build -t peanut:custom .
mkdir /data/peanut
echo -e "WEB_HOST:localhost\nWEB_PORT: 3493\nWEB_HOST: 0.0.0.0\nWEB_PORT: 8080\nUSER: admin\nPASSWORD: secret" > /data/peanut/settings.yml
docker run -d --name peanut --network monitoring --restart unless-stopped -p 8080:8080 -v /data/peanut:/config peanut:custom
docker logs peanut
```
```
> peanut@3.0.0-beta.0 start
> HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js

  ▲ Next.js 14.2.13
  - Local:        http://localhost:8080
  - Network:      http://0.0.0.0:8080

 ✓ Starting...
 ✓ Ready in 90ms
```

Brandawg93, 2024-10-21:
Does your UPS require that you manually specify the serial number in the configuration files? I've never seen that before. I tried replicating your setup and mine breaks entirely when adding the serial to the configuration.

dasilvatek, 2024-10-21:
@Brandawg93 I just tried removing the serial number from NUT ups.conf and everything still works, so I presume it's not required. That said, the same issue persists - only the first digit shows via the same methods in PeaNUT.

Brandawg93, 2024-10-21:
Interesting. That value can be either a string or number so it shouldn’t be a type issue from the peanut side of things. Could you login directly to the nut server and print all variables and see what it gives you? I’m wondering if there is some sort of delimiter in the value e.g. “-“.

dasilvatek, 2024-10-21:
@Brandawg93 the parameter type is defined as NUMBER for all instances (perhaps it should be string?):
```
~$curl -X GET http://localhost:8888/api/v1/devices/myups/var/ups.serial/type
"NUMBER"
~$ curl -X GET http://localhost:8888/api/v1/devices/myups/var/device.serial/type
"NUMBER"
```

Here is the output for all parameters on NUT:
```
~$ upsc myups@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.date: 2001/09/25
battery.mfr.date: 2017/02/18
battery.runtime: 15750
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.2
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Back-UPS XS 1500G
device.serial: 4B1707P21318  
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.productid: 0002
driver.parameter.synchronous: auto
driver.parameter.vendorid: 051d
driver.state: quiet
driver.version: 2.8.1
driver.version.data: APC HID 0.100
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.sensitivity: medium
input.transfer.high: 139
input.transfer.low: 88
input.voltage: 122.0
input.voltage.nominal: 120
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.firmware: 866.L9 .D
ups.firmware.aux: L9 
ups.load: 0
ups.mfr: American Power Conversion
ups.mfr.date: 2017/02/18
ups.model: Back-UPS XS 1500G
ups.productid: 0002
ups.realpower.nominal: 865
ups.serial: 4B1707P21318  
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

Brandawg93, 2024-10-21:
Ah. That'll do it. I guess I can't trust the driver to always give me accurate information. I've added a check for strings and numbers and made a new [test version](https://hub.docker.com/layers/brandawg93/peanut/test/images/sha256-a5454f9916fd5ce3a501f4f0380367ddac22ff3dd399e486b8de6b540762756e?context=repo). Let me know if it works for you!

dasilvatek, 2024-10-21:
@Brandawg93 yes sir that works! I look forward to your latest official release.
[image]

Brandawg93, 2024-10-21:
Awesome! The latest version has just been released. Let me know if you run into any issues!

closed 2024-10-21 by Brandawg93