Brandawg93/PeaNUT #75: [Feature Request]: Add an option to display load as wattage instead of percentage of capacity

7eventy7, 2024-10-19:
### Enhancement of issue

It would be nice to see the current load as wattage instead of what we currently have which is just a percentage.

### Solution

Maybe a configuration option to change between the two options?

dasilvatek, 2024-10-22:
Do you have this data point reporting from your UPS?

I don't see it coming directly from my UPS. You would have to do something creative like taking the **ups.load** percentage from the **ups.realpower.nominal** total wattage. For example, I have 2 routers and 1 USB drive currently plugged into the UPS for a load of 3% on a nominal load of 865W, so the current load in wattage is about 26W.
[image]
[image]

7eventy7, 2024-10-22:
Yes, NUT reports the following in the unraid dashboard:

USAGE% / WATTAGE

7eventy7, 2024-10-22:
[image]

7eventy7, 2024-10-29:
Bump as this is my top feature request

Brandawg93, 2024-10-30:
Could you try out the latest [test version](https://hub.docker.com/layers/brandawg93/peanut/test/images/sha256-209a67cb9f13fc0125dcfc12d084e60c7755d110d2b8d5c3a994755e98f1e242?context=repo)?

Simply click the current load card.

7eventy7, 2024-10-30:
> Could you try out the latest [test version](https://hub.docker.com/layers/brandawg93/peanut/test/images/sha256-209a67cb9f13fc0125dcfc12d084e60c7755d110d2b8d5c3a994755e98f1e242?context=repo)?
> 
> Simply click the current load card.

Sorry for the delay, seems to be working great. A new API key/value should do the trick! Thanks for the super quick response!

Brandawg93, 2024-10-30:
Awesome! I'll make the UI changes in the next release. As far as the API, everything received comes directly from the nut server. I'd prefer not to make any exceptions to that with custom key/values.

7eventy7, 2024-10-30:
> Awesome! I'll make the UI changes in the next release. As far as the API, everything received comes directly from the nut server. I'd prefer not to make any exceptions to that with custom key/values.

Oh rats is there any potential workaround? That API stat is really what I'm after as I use peanut in my integrated Homepage dashboard,
[image]

7eventy7, 2024-10-30:
> Oh rats is there any potential workaround? That API stat is really what I'm after as I use peanut in my integrated Homepage dashboard, [...]

Edit: Trying to think of a workaround made me inspect the code a bit closer and it looks like I convert the % to a W value by knowing the total capacity of my UPS!

7eventy7, 2024-10-30:
> Awesome! I'll make the UI changes in the next release. As far as the API, everything received comes directly from the nut server. I'd prefer not to make any exceptions to that with custom [...]

![image](https://private-user-images.githubusercontent.com/75962770/381661733-22c03795-0091-4356-8178-2e1566b4de99.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzMDg5NTIsIm5iZiI6MTczMDMwODY1MiwicGF0aCI6Ii83NTk2Mjc3MC8zODE2NjE3MzMtMjJjMDM3OTUtMDA5MS00MzU2LTgxNzgtMmUxNTY2YjRkZTk5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMwVDE3MTczMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM1Njc2YjU0YTkzYmRmZjBmMjY2YzZlNzU5MDk1OTQ5OTNmOWU0Yzg4NjBiODlmZTFlYWU1OWRiOTAwMTViN2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.oVz8Zzedz_zz4WNF8MHAa4KNzXQJHgntR1rXGYbC1So)

7eventy7, 2024-10-30:
Trying to think of a workaround made me inspect the code a bit closer and it looks like I convert the % to a W value by knowing the total capacity of my UPS!

```              remap:
                - field: ups.load
                  op: multiply
                  value: 6
            - field: battery.runtime
              label: Runtime
              format: number
              scale: 1/60
              suffix: "mins"
            - field: ups.status
              label: Status
              format: text
              remap:
                - value: OL
                  to: Online
                - value: OL CHRG
                  to: Online
                - value: OB
                  to: On Battery
                - value: LB
                  to: Low Battery
                - any: true
                  to: Unknown```

Brandawg93, 2024-10-30:
Ah yes! A custom widget would totally do the trick! I should create a wiki of these little tricks here…🤔 Maybe a homepage custom widgets wiki.

Brandawg93, 2024-10-31:
This is now added in v3.3.0.

closed 2024-10-31 by Brandawg93