Brandawg93/PeaNUT #78: [Bug]: settings.yml

spocoeric, 2024-10-21:
### Describe the bug

I don.t see settings.yml anywhere in the source code. Where is this file located? I'm getting Internal Server Error when upgrading to 3.0.0, and I suspect I need to update some settings there to correct.

### Debug Output

*

### Steps to reproduce

1.  Searched provided zip of Source Code

### Device Type

Proxmox

### UPS Device

CPS ST Series

### Last Working Version

2.60

Brandawg93, 2024-10-21:
It’s not in the source code. It’s generated at runtime on the first run. Take a look at the examples and you can see how to gain access to it.

dasilvatek, 2024-10-21:
@spocoeric I mounted a volume to store the configuration file:
`docker run -d --name peanut -v /data/peanut/config:/config brandawg93/peanut`

Running your previous docker run for v2.x will create and populate the variables, or you can create a settings.yml file too - either way works. Thanks @Brandawg93 you rock!

Brandawg93, 2024-10-22:
I just noticed I forgot to update the docker run command example. Thanks for the catch! Glad it's working for you now.

closed 2024-10-22 by Brandawg93

spocoeric, 2024-10-22:
This isn't working for me, possibly because this is an LXC install on Proxmox, and not docker?

Brandawg93, 2024-10-22:
Possibly. I have no idea what LXC is but docker is the expected deploy method.

spocoeric, 2024-10-22:
if I had an example settings.yml, I believe I know were it should reside and could try it, but without it generating, I don't know how it is formatted.

spocoeric, 2024-10-22:
Working fine with docker. I will play around with it. Thanks.