Brandawg93/PeaNUT #82: [Bug]: Internal Server Error after updating container

7eventy7, 2024-10-23:
### Describe the bug

Internal Server Error when trying to access the webUI after updating the container automatically last night.

### Debug Output

TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
    at o.loadFromEnvVars (/.next/server/chunks/161.js:1:573)
    at o.initWithEnvVars (/.next/server/chunks/161.js:1:787)
    at Module.em (/.next/server/instrumentation.js:5:10240)
    at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.0_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
    at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.0_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)

### Steps to reproduce

Not sure

### Device Type

Unraid - Docker

### UPS Device

APC Back-UPS XS 1050M - usbhid-ups

dasilvatek, 2024-10-23:
can you share your docker run or compose?

armond-avanes, 2024-10-23:
I have the exact same issue here, with no mapped `settings.yml` and only using env variables.

pidzej, 2024-10-23:
I have the exact same issue here, with no mapped settings.yml and only
using env variables.

Brandawg93, 2024-10-23:
I’m working on this. Check back soon!

Brandawg93, 2024-10-23:
Could someone please test my latest test version? If it works, I'll push an update for everyone else.

[test version](https://hub.docker.com/layers/brandawg93/peanut/test/images/sha256-60e13a9a2fca8c031c3bee73926ec17273699f68fab4864ea3a1242e65d646c3?context=repo)
https://github.com/Brandawg93/PeaNUT/pull/83

7eventy7, 2024-10-23:
> Could someone please test my latest test version? If it works, I'll push an update for everyone else.
> 
> [test version](https://hub.docker.com/layers/brandawg93/peanut/test/images/sha256-60e13a9a2fca8c031c3bee73926ec17273699f68fab4864ea3a1242e65d646c3?context=repo) #83

Switched repo to :test. Seems to have fixed the issue. Any chance you could add my feature request too? :p

Brandawg93, 2024-10-23:
Awesome! And I’m still looking into UI customization like that. For now, you could use the persistent data example to build a grafana graph with that feature.

Brandawg93, 2024-10-23:
Fixed in v3.1.2!