Brandawg93/PeaNUT #84: [Bug]: Internal server error due to no settings.yml file

jzucker2, 2024-10-23:
### Describe the bug

The previous issue was marked as resolved but did not fix this issue:
https://github.com/Brandawg93/PeaNUT/issues/82

### Debug Output

```
peanut-schrader  | 
peanut-schrader  | > peanut@3.1.2-test.0 start
peanut-schrader  | > HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js
peanut-schrader  | 
peanut-schrader  |   ▲ Next.js 15.0.1
peanut-schrader  |   - Local:        http://localhost:8080
peanut-schrader  |   - Network:      http://0.0.0.0:8080
peanut-schrader  | 
peanut-schrader  |  ✓ Starting...
peanut-schrader  |  ✓ Ready in 412ms
peanut-schrader  | Failed to prepare server TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
peanut-schrader  | Failed to prepare server TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
peanut-schrader  | TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
peanut-schrader  | TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
peanut-schrader  | TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
peanut-schrader  | TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
peanut-schrader  | TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
peanut-schrader  | TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
peanut-schrader  | TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
[... 24 lines cut ...]
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
peanut-schrader  | TypeError: Cannot read properties of undefined (reading 'NUT_HOST')
peanut-schrader  |     at l.loadFromEnvVars (/.next/server/chunks/921.js:1:651)
peanut-schrader  |     at l.initWithEnvVars (/.next/server/chunks/921.js:1:863)
peanut-schrader  |     at Module.em (/.next/server/instrumentation.js:5:10240)
peanut-schrader  |     at async NextNodeServer.runInstrumentationHookIfAvailable (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:493:9)
peanut-schrader  |     at async NextNodeServer.prepareImpl (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/server/next-server.js:487:9)
```

### Steps to reproduce

1. Run the app without a settings.yml file (see previous issue for more details)

### Device Type

Any device

### UPS Device

Any device

### Last Working Version

2.6.1

Brandawg93, 2024-10-23:
And this is with the latest v3.1.2? I just tried on my production server and it works fine.

jzucker2, 2024-10-23:
I'm pretty sure the bug is from new settings parsing code introduced in the recent jump to v3

https://github.com/Brandawg93/PeaNUT/blob/69474ca1ed40d893157583b6b25b09d5fba315f4/src/server/settings.ts#L37-L48

```
  private loadFromEnvVars(): void {
    for (const key of Object.keys(ISettings)) {
      if (!this.data[key] && process.env[key]) {
        const eVar = key as keyof typeof ISettings
        if (typeof ISettings[eVar] === 'number') {
          this.data[key] = parseInt(process.env[key], 10)
        } else {
          this.data[key] = process.env[key]
        }
      }
    }
  }
```

In the case of only having env vars, this would be an empty dict and would fail with a default `NUT_HOST: undefined,`. It might be a quick hack to do `NUT_HOST: "foo",` or there could be a little more validation.

I'm curious how you tested this? If you just update an existing server that would likely already have a valid settings.yml file, right?

Brandawg93, 2024-10-23:
I'm testing by removing all volumes from the peanut container. This will force it to only create the settings.yml inside the container making the file not persist between containers.

Your specific error says that `this.data` is undefined, not `this.data['NUT_HOST']`. But `this.data` is initialized to `{}` so I'm not really sure why it's throwing this error.

https://github.com/Brandawg93/PeaNUT/blob/main/src/server/settings.ts#L25

jzucker2, 2024-10-23:
I got it working by removing the volume. So if you provide a volume but **no settings.yml file** and use the env vars, it will crash.

Seems like the docs might need updating or a fix could be helpful.

The 2 working states:

1) Provide a volume **and** a settings.yml file
2) Use env vars with **no** settings.yml file

jzucker2, 2024-10-23:
Thanks for moving so quickly on this btw!

Brandawg93, 2024-10-23:
Actually, I think you have a valid point. Looking at the yaml load function, it will return undefined if it doesn't understand something. I've created a new test version and made a change to fix it. Do you think you could test your setup like you had it before to see if it works now?

[Brandawg93/peanut:test](https://hub.docker.com/layers/brandawg93/peanut/test/images/sha256-03dae6db83cfd600437441abcf41491a48da7dcc59458ef9d5b777118d8ce556?context=repo)
https://github.com/Brandawg93/PeaNUT/pull/85

jzucker2, 2024-10-23:
> Actually, I think you have a valid point. Looking at the yaml load function, it will return undefined if it doesn't understand something. I've created a new test version and made a change to fix it. [...]

Yea! It might take me a little bit of time, I should be able to get this tonight or tomorrow. Thanks so much!

jzucker2, 2024-10-23:
It loads but doesn't seem happy:

```
peanut-schrader  | 
peanut-schrader  | > peanut@3.1.2 start
peanut-schrader  | > HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js
peanut-schrader  | 
peanut-schrader  |   ▲ Next.js 15.0.1
peanut-schrader  |   - Local:        http://localhost:8080
peanut-schrader  |   - Network:      http://0.0.0.0:8080
peanut-schrader  | 
peanut-schrader  |  ✓ Starting...
peanut-schrader  |  ✓ Ready in 421ms
peanut-schrader  | Error: Failed to find Server Action "8f4ce8714594e363af2703037254304c2b696dd1". This request might be from an older or newer deployment. Original error: Cannot read properties of undefined (reading 'workers')
peanut-schrader  |     at rf (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:127:2692)
peanut-schrader  |     at /node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:127:757
peanut-schrader  |     at AsyncLocalStorage.run (node:internal/async_local_storage/async_hooks:91:14)
peanut-schrader  |     at rd (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:126:11732)
peanut-schrader  |     at n8 (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:132:1243)
peanut-schrader  |     at /node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:132:6404
peanut-schrader  |     at AsyncLocalStorage.run (node:internal/async_local_storage/async_hooks:91:14)
peanut-schrader  |     at tm (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:124:48126)
peanut-schrader  |     at /node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:132:6254
peanut-schrader  |     at AsyncLocalStorage.run (node:internal/async_local_storage/async_hooks:91:14)
peanut-schrader  | Error: Failed to find Server Action "40a33b5e8ea8d865f1541a735e21223e572e47ae". This request might be from an older or newer deployment. Original error: Cannot read properties of undefined (reading 'workers')
peanut-schrader  |     at rf (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:127:2692)
peanut-schrader  |     at /node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:127:757
peanut-schrader  |     at AsyncLocalStorage.run (node:internal/async_local_storage/async_hooks:91:14)
peanut-schrader  |     at rd (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:126:11732)
peanut-schrader  |     at n8 (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:132:1243)
peanut-schrader  |     at /node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:132:6404
peanut-schrader  |     at AsyncLocalStorage.run (node:internal/async_local_storage/async_hooks:91:14)
peanut-schrader  |     at tm (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:124:48126)
peanut-schrader  |     at /node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:132:6254
peanut-schrader  |     at AsyncLocalStorage.run (node:internal/async_local_storage/async_hooks:91:14)
```

Brandawg93, 2024-10-23:
That seems to be some sort of caching issue. At least the original issue is fixed! I'll release another patch soon. Thanks for helping debug this.

Brandawg93, 2024-10-23:
Fixed in v3.1.3. Just let me know if you run into any other issues.

closed 2024-10-23 by Brandawg93