Brandawg93/PeaNUT #90: [Bug]: Internal Server Error (Failed to find Server Action)

will6789, 2024-10-24:
### Describe the bug

After updating to v3, I'm getting an Internal Server Error when trying to access the WebUI

Here's my compose file (only change since updating to v3 was adding the volume):
```
services:
  peanut:
    image: brandawg93/peanut:latest
    container_name: PeaNUT
    restart: unless-stopped
    volumes:
      - peanut_config:/config
    ports:
      - 8080:8080
    environment:
      - NUT_HOST=192.168.1.201
      - NUT_PORT=3493
      - WEB_PORT=8080
      - USERNAME=monuser

volumes:
  peanut_config:
```

### Debug Output

```
> peanut@3.1.3 start
> HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js
  ▲ Next.js 15.0.1
  - Local:        http://localhost:8080
  - Network:      http://0.0.0.0:8080
 ✓ Starting...
 ✓ Ready in 594ms
Error: Failed to find Server Action "4ad7d1a6980d90067b44098e1f9981625c9d9887". This request might be from an older or newer deployment. Original error: Cannot read properties of undefined (reading 'workers')
    at rf (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:127:2692)
    at /node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:127:757
    at AsyncLocalStorage.run (node:internal/async_local_storage/async_hooks:91:14)
    at rd (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:126:11732)
    at n8 (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:132:1243)
    at /node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:132:6404
    at AsyncLocalStorage.run (node:internal/async_local_storage/async_hooks:91:14)
    at tm (/node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:124:48126)
    at /node_modules/.pnpm/next@15.0.1_@babel+core@7.23.6_@playwright+test@1.48.1_react-dom@18.3.1_react@18.3.1__react@18.3.1/node_modules/next/dist/compiled/next-server/app-page.runtime.prod.js:132:6254
    at AsyncLocalStorage.run (node:internal/async_local_storage/async_hooks:91:14)
```

### Steps to reproduce

1. Run PeaNUT and try to access the interface

### Device Type

Synology DS920+

### UPS Device

Eaton 3S 850 - usbhid-ups

### Last Working Version

2.6.1

Brandawg93, 2024-10-25:
This should be fixed in the latest [test build](https://hub.docker.com/layers/brandawg93/peanut/test/images/sha256-9bb4a36f1897ada2448354fd3998aa4fb1aa88d46e92d6d917c7f84f717666af?context=repo). Let me know if this works for you!

will6789, 2024-10-25:
Just tried it, I'm still getting the Internal Server Error from the UI but I'm no longer getting any errors in the container log. (I'm aware that's probably not much help aha)

Brandawg93, 2024-10-25:
It may be a configuration issue if there are no other logs. Are you able to log out and log back in?

will6789, 2024-10-25:
Yeah, when I click the log out button from the error screen it takes me to the form to configure the address and port. After entering these, clicking the test button gives me a green tick (which I'm assuming means it can connect to the host ok) but then clicking connect returns another Internal Server Error and a new error in the log:
```
Error: Timeout
    at Timeout._onTimeout (/.next/server/app/api/v1/devices/[device]/clients/route.js:2:142)
    at listOnTimeout (node:internal/timers:594:17)
    at process.processTimers (node:internal/timers:529:7)
```

gaborho, 2024-10-26:
Hi @will6789 

It seems you would need to use network_mode: host in your Docker Compose file. At least this is how I could make it running on my Synology DS224+ on DSM 7.2.2-72806

Also, defining USERNAME=monuser environment variable is not necessary.

Here is my working Docker Compose file:

``
services:
  peanut:
    image: brandawg93/peanut:latest
    container_name: peanut
    restart: unless-stopped
    volumes:
      - /volume1/docker/peanut:/config
    environment:
      - NUT_HOST=localhost
      - NUT_PORT=3493
      - WEB_PORT=8484
    network_mode: host
``

You will be able to access it on the following address: <synology_ip_address>:8484

will6789, 2024-10-26:
Ok, I've figured it out! The Synology NUT server has an IP whitelist and refuses to talk to anything not on the list. Previously, I'd whitelisted the IP of the container but it must've changed when the container was re-created to update to v3. 
Even when using @gaborho's config I still had to add the NAS's own IP to the whitelist (since the requests are now coming from the host IP) or I still got the timeout error. 

Looks like `network_mode: host` is the best way of doing it in this case, else you'd end up manually updating that whitelist everytime the app is updated.

Thanks to both of you for your help!

closed 2024-10-26 by will6789