Brandawg93/PeaNUT #92: Is InfluxDB required? Prometheus support?

0dragosh, 2024-10-26:
### Enhancement of issue

Are the InfluxDB env vars required or does this function standalone? it isn’t clear from the README.

Also is Prometheus supported or are there any plans to introduce Prometheus metrics support?

Thanks!

### Solution

Find out whether the InfluxDB env vars are required or optional and whether prometheus is supported. If not can we put it on the roadmap.

dasilvatek, 2024-10-27:
InfluxDB is not required. I don’t have it setup and Peanut works just fine.

ihatethecloud, 2024-12-01:
+1 prometheus support

SR-G, 2024-12-29:
Same, i would definitely be interested by a prometheus export.
In fact the tool should (ideally ...) be about : 
- a daemon, collecting UPS metrics
- a "prometheus exporter" endpoint ("/metrics" path), maybe disabled by default and enable-able through configuration or environment variable
- a "web UI" (the current one), ideally enabled by default but disable-able through configuration of environment variable (for whoever - like me ! - only interested by the prometheus exporter + grafana dashboard (this is where i collect/maintain all my dashboards))

Brandawg93, 2025-02-08:
I'm finally getting around to this and am working on #170! I've published a test build with this turned on. Could someone try `brandawg93/peanut:test` and report back?

SR-G, 2025-02-08:
Nice
Is there already a docker image (tag) available somewhere ?
(was willing to compile from source from the PR, but it's failing on my side at pnpm level)

Brandawg93, 2025-02-08:
Yes, it should be available at the image tag I posted above.

Brandawg93, 2025-02-09:
I've also added a new [Grafana dashboard](https://github.com/Brandawg93/PeaNUT/tree/prometheus/examples/prometheus/grafana/provisioning/dashboards) example if that helps anyone get up and running quicker.

ihatethecloud, 2025-02-09:
Worked nice. The only quirk I found was with the grafana dashboard. I had to refresh each panel to get the data in.  I’ll leave it for a few hours and come back with feedback.

Thanks

Brandawg93, 2025-02-09:
Nice! I updated the dashboard to automatically refresh every 10 seconds.