Brandawg93/PeaNUT #94: [Bug]: Failed connection test still displays green check mark

remz1337, 2024-10-27:
### Describe the bug

When setting up Peanut for the first time, the form where it asks for server and port has a test button. I made a mistake and put the wrong port number. I clicked the test button and it still showed the green check mark, so I was assuming it was fine, but then when I clicked connect, it showed the "internal error" page.

Also, I would recommend to specify on this form that the values are for the NUT server, and pre-fill the port value with default port (3493).

### Debug Output

The test function should check for the actual connection status, as I can see in the logs that the connection was refused.
```
Oct 27 14:12:27 peanut peanut[941]:   ▲ Next.js 15.0.1
Oct 27 14:12:27 peanut peanut[941]:   - Local:        http://localhost:3000
Oct 27 14:12:27 peanut peanut[941]:   - Network:      http://0.0.0.0:3000
Oct 27 14:12:27 peanut peanut[941]:  ✓ Starting...
Oct 27 14:12:27 peanut peanut[941]:  ✓ Ready in 83ms
Oct 27 14:14:25 peanut peanut[941]: Error: connect ECONNREFUSED 192.168.1.123:3943
Oct 27 14:14:25 peanut peanut[941]:     at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1607:16)
Oct 27 14:14:25 peanut peanut[941]:     at TCPConnectWrap.callbackTrampoline (node:internal/async_hooks:130:17) {
Oct 27 14:14:25 peanut peanut[941]:   errno: -111,
Oct 27 14:14:25 peanut peanut[941]:   code: 'ECONNREFUSED',
Oct 27 14:14:25 peanut peanut[941]:   syscall: 'connect',
Oct 27 14:14:25 peanut peanut[941]:   address: '192.168.1.123',
Oct 27 14:14:25 peanut peanut[941]:   port: 3943
Oct 27 14:14:25 peanut peanut[941]: }
```

### Steps to reproduce

1. Fresh install
2. Enter correct NUT server address
3. Enter wrong NUT server port
4. Click "Test"

### Device Type

Proxmox server

### UPS Device

USB

dasilvatek, 2024-10-27:
You can run NUT on any port you want as configured in upsd.conf, so it would be inappropriate to lock this value in Peanut.
```
LISTEN 127.0.0.1 3493
LISTEN ::1 3493
```

remz1337, 2024-10-28:
yes, that was just a suggestion. I think it's common good UX practice to prefill when there's a default or common accepted standard. User can still overwrite it.

Issue remains though. Test button should check for a successful connection, not just ping the IP

Brandawg93, 2024-10-28:
Thanks for the bug report! There was definitely an issue. I've fixed it so it will actually tell you the issue now. I also went ahead and added 3493 as the default port. Obviously, this can be changed if the user needs.

Please try the latest test version and let me know if it works for you!

https://hub.docker.com/layers/brandawg93/peanut/test/images/sha256-32aaec3235d96cce65156119cfeb3fde3e62a6ef323789703bf0a2036ad2f125?context=repo

remz1337, 2024-10-28:
I don't have a docker environment, but I tested by cloning the more-optimizations branch. Issue still there. I actually pushed it further by inputting random values in both fields and the test still show the green checkmark

[image]

Brandawg93, 2024-10-28:
It should just be https://github.com/Brandawg93/PeaNUT/pull/93.

remz1337, 2024-10-28:
Ok I was able to test properly and everything works as expected. Thank you!

Brandawg93, 2024-10-28:
Awesome! This is now live in v3.2.1. Thanks for testing!

closed 2024-10-28 by Brandawg93