Brandawg93/PeaNUT #109: [Feature Request]: Add custom names for UPS

tOOmki, 2024-12-01:
### Enhancement of issue

I have several of the same UPS.
In the drop-down list, they all have the same description.

### Solution

Add custom names for UPS

elfensky, 2024-12-13:
Can you not fix that by providing a - NAME variable to the upsd service?

Brandawg93, 2024-12-13:
We briefly discussed it in another issue. They are using Synology which doesn't allow to change the name. But I do hesitate to implement this considering it should be done on the server side.

voltaweb-machines, 2024-12-13:
Could be as simple as an additional "alias" field that gets priority if it has content. But I do understand your point too.

Brandawg93, 2025-02-18:
Looks like someone found how to change the names in #184! I'm going to close this for now, since I don't plan on implementing it. I may revisit in the future.

tracerrx, 2026-03-31:
+1 for either displaying/importing the description field from the server or allowing us to define one.

gobbledygoober, 2026-08-16:
I am writing to also suggest the ability to customize the Description column in the user interface. I have two similar NUT servers that use different UPS models, and you can only distinguish them if you click on the Details button in each row:

[image]

Thanks for your work on this project.