Brandawg93/PeaNUT #111: [Bug]:

StoicaRemus, 2024-12-02:
### Describe the bug

Changes to INFLUX_INTERVAL variable not changing the interval for Influx ingestion.
Using docker, I've noticed that INFLUX_INTERVAL changes do not reflect the set variable, I have tried with lower than 10, higher than 10, even with 60s, the interval remains 10s each time. Modifying other ENVs are reflected after container deployment, checked via webui. Also there is a typo 'Inverval'

### Debug Output

No logs displayed.

### Steps to reproduce

1. Use any value for INFLUX_INTERVAL
2. Verify InfluxDB

### Device Type

Ubuntu dockerized

### UPS Device

Eaton 850 Elipse

Brandawg93, 2024-12-02:
You may need to change the value in the configuration file as well. But I can also make that configurable through the UI. I'll probably do that on the next release.

Brandawg93, 2024-12-02:
#112

Brandawg93, 2024-12-05:
This is now available in v4.1.0!

StoicaRemus, 2024-12-05:
i can modify the interval now in the settings, but it resets back to 10s each time.

Brandawg93, 2024-12-06:
Each time you refresh the page or restart the app? Also, do you have a volume defined for your configuration file?

StoicaRemus, 2024-12-06:
How can i reproduce it each time:
Update the Influx Interval from the settings
Click apply
Refresh Page

Value reverts back to default (10s)
If I try to update by first backspacing the value, it will autofill a 0 each time. I can update the bucket name etc and it will remain. No volumes binded to the container, removed the ENV vars related to influx, still same behaviour.

Brandawg93, 2024-12-07:
I have a fix for this. 7861a91815e6670b0f6c88c8ec9c87a102b57ee2
Will add it to the next release.

Brandawg93, 2024-12-07:
This should be fixed in the latest release (v4.2.0). Just let me know if it's still an issue for you.

closed 2024-12-07 by Brandawg93