Brandawg93/PeaNUT #113: [Bug]: The contents of the settings.yml file are gone and the default settings are back

tOOmki, 2024-12-03:
### Describe the bug

I go to the page and I get this message:
"No devices found on this server."

The contents of the settings.yml file are gone and the default settings are back. 
This is the third time this has happened.

I did not do any restart.
Version 4.0.2

### Debug Output

.Where can I find the logs ?

### Steps to reproduce

1. none

### Device Type

Proxmox 8.2.4

### UPS Device

APC Back-UPS CS 350, APC Back-UPS CS 650, CyberPower UT850EG

Brandawg93, 2024-12-03:
This may be fixed in #115. Could you try out the latest test version and let me know if it's working better for you?

`brandawg93/peanut:test`

tOOmki, 2024-12-04:
I use https://community-scripts.github.io/ProxmoxVE/ for installation
I don't know how to change the version to "better settings".

tOOmki, 2024-12-04:
This is what the command to update the LXC container looks like.

The update does not go through because there is no next version"
"✓ No update required. PeaNUT is already at v4.0.2"

bash -c "$(wget -qLO - https://github.com/community-scripts/ProxmoxVE/raw/main/ct/peanut.sh)"

Brandawg93, 2024-12-04:
Hm. That's above me. Maybe someone in that community could help more. I see where the script is getting the latest release version from GitHub and pulling that from docker. But that doesn't really make any sense considering you can just get tags from docker directly. Either way, I will probably just release the work on a new version soon. I don't think the changes could really hurt, only help.

tOOmki, 2024-12-04:
They are working on selecting the appropriate branch or version. But when will it work???
https://github.com/community-scripts/ProxmoxVE/pull/330

Brandawg93, 2024-12-05:
Let me know if v4.1.0 works better for you.

Brandawg93, 2024-12-07:
This should be fixed in the latest release (v4.2.0). Just let me know if it's still an issue for you.

closed 2024-12-07 by Brandawg93

tOOmki, 2024-12-08:
I installed fresh version  4.2
The problem is recurring.

But I think I know why, in my case I have 3 NAS from Synology. I have a work schedule set, the moment the NAS is turned off the settings are cleared. It does not always happen and I do not know why.