Brandawg93/PeaNUT #119: [Bug]: /api/v1/devices/[device] throws 500 on all devices not on the first configured nut server

sjafferali, 2024-12-05:
### Describe the bug

After successfully configuring multiple nut servers in v4.1.0, the UI shows the UPS devices on both servers without issues, however, when attempting to query the API for any devices that is not on the first nut server configured results in a 500 error. 

### Debug Output

The log when this happens does show timeouts. 

```
 ⨯ Error: Timeout
    at Timeout._onTimeout (/.next/server/chunks/236.js:1:5450)
 ⨯ Error: Timeout
    at Timeout._onTimeout (/.next/server/chunks/236.js:1:5450)
 ⨯ Error: Timeout
    at Timeout._onTimeout (/.next/server/chunks/236.js:1:5450)
 ⨯ Error: Timeout
    at Timeout._onTimeout (/.next/server/chunks/236.js:1:5450)
```

I assume this endpoint has not been updated to support multiple nut servers which was new functionality added in v4.0.0, which likely means it may be attempting to query the first nut server for the device that is actually located on the second nut server configured. 

### Steps to reproduce

1. Configure two different nut servers, each setup with a single device. 
2. Hit the API endpoint /api/v1/devices/[first device] and see a working response.
3. Attempt to hit the API endpoint again for the second device and the result is a 500 error. 

### Device Type

Docker on home server

### UPS Device

SMART1000RM1U - snmp

Brandawg93, 2024-12-07:
This should be fixed in the latest release (v4.2.0). Just let me know if it's still an issue for you.

closed 2024-12-07 by Brandawg93

sjafferali, 2024-12-08:
@Brandawg93 Thanks. Works great now!