Brandawg93/PeaNUT #125: [Bug]: Image got too big

vladaurosh, 2024-12-15:
### Describe the bug

Just realised that uncompressed docker image is 685MB (arm64). :o
Culprit seems to be .next/cache/webpack which is 395MB. 
Adding 
`&& rm -rf .next/standalone/.next/cache/webpack`
at the end of 
`RUN npm install -g pnpm && npm run telemetry && npm run build`
got the image down to 270MB.

Can this be added in Dockerfile?

### Debug Output

N/A

### Steps to reproduce

N/A

### Device Type

rpi 5

### UPS Device

Eaton 5P

Brandawg93, 2024-12-15:
Interesting. I went ahead and removed the entire cache folder and republished over latest. It seems to have reduced the build size a bit. Let me know if you see that too.

closed 2024-12-16 by Brandawg93

vladaurosh, 2024-12-16:
Looking good now, 685MB -> 269MB. Thanks.