Brandawg93/PeaNUT #126: [Bug]: A lot of DNS queries comming from PeaNUT

vladaurosh, 2024-12-18:
### Describe the bug

I got the error message in UI, saying Internal server error, log is attached.
I am using hostname instead of IP address for NUT_HOST, my local dns is Pihole and in Pihole I got a message that host has been rate limited (up to 1000 requests in a minute). It's the host running peanut container. Indeed Pihole log is showing a lot of dns requests coming from the host where peanut is running. 
Anything can be done to avoid this?

### Debug Output

⨯ Error: Connection failed: getaddrinfo EAI_AGAIN <nut_server_hostname>
    at s.getCommand (.next/server/chunks/66.js:2:883)
    at async s.getVarDescription (.next/server/chunks/66.js:3:1486)
    at async (.next/server/chunks/66.js:3:214)
    at async s.getData (.next/server/chunks/66.js:3:402)
    at async (.next/server/app/api/v1/devices/route.js:1:1420)
    at async (.next/server/app/api/v1/devices/route.js:1:1480)

### Steps to reproduce

n/a

### Device Type

rpi 5

### UPS Device

Eaton 5P

### Last Working Version

4.2.1

rmlamarche, 2024-12-24:
Possibly related, I'm seeing a burst of 92 separate tcp connections to my NUT server every 10 seconds with v4.2.1, and I'm also seeing internal server error messages in the UI.

Brandawg93, 2024-12-26:
The fix for this is now live in v4.3.0.

closed 2024-12-26 by Brandawg93