Brandawg93/PeaNUT #131: [Bug]: Can't connect to influxdb2 on same docker network

schoolbust, 2024-12-25:
### Describe the bug

Trying to connect peanut to an influxdb2 container on the same docker network.

Here is my docker compose;

  peanut:
    image: brandawg93/peanut:latest
    container_name: PeaNUT
    networks:
      peanut:
        ipv4_address: 192.168.207.30
    restart: unless-stopped
    volumes:
      - $APPDATA/peanut:/config
    ports:
      - 9999:8080
    environment:
      WEB_PORT: 8080
      PUID: $PUID
      PGID: $PGID
      TZ: $TZ

  influxdb2-peanut:
    image: influxdb:2
    container_name: influxdb-PeaNUT
    hostname: influxdb2-peanut
    volumes:
      - $APPDATA/influxdb2-peanut/db:/var/lib/influxdb2
      - $APPDATA/influxdb2-peanut/config:/etc/influxdb2
    networks:
      peanut:
        ipv4_address: 192.168.207.35
    ports:
      - 8087:8086
    environment:
      DOCKER_INFLUXDB_INIT_MODE: setup
      DOCKER_INFLUXDB_INIT_USERNAME: Admin
      DOCKER_INFLUXDB_INIT_PASSWORD: $NUTPASS
      DOCKER_INFLUXDB_INIT_ORG: homelab
      DOCKER_INFLUXDB_INIT_BUCKET: peanut
    restart: unless-stopped

### Debug Output

Error writing to InfluxDB: Error: Unsupported protocol "null in URL: "192.168.207.35"
    at new ed (.next/server/chunks/749.js:4:2693)
    at new ey (.next/server/chunks/749.js:4:9114)
    at new ev (.next/server/chunks/749.js:4:10245)
    at Timeout._onTimeout (.next/server/chunks/749.js:6:5453)

### Steps to reproduce

n/a

### Device Type

debian virtual machine under proxmox

### UPS Device

n/a

Brandawg93, 2024-12-25:
I think you need to prepend “http://“ to your server address.

schoolbust, 2024-12-25:
unfortunately still getting the same error with that :(

Brandawg93, 2024-12-26:
It seems that the influx server isn't allowing connections. Peanut is able to queue 129 items to write, but the server will not accept them.

schoolbust, 2024-12-26:
wondering if there's a certain influxdb v2 version that's recommended? Do you by chance have a recommended image that you've confirmed working?

Brandawg93, 2024-12-27:
Check out the examples in the repo. I have a [docker-compose.yml](https://github.com/Brandawg93/PeaNUT/blob/main/examples/persistant-data/docker-compose.yml) that has a working image along with pre-made grafana dashboards.

schoolbust, 2024-12-27:
ah didn't see that docker-compose.yml file before! apologies but now I got it all to work, looks like I was just really missing the environment callouts. Thanks so much closing due to my own error.

closed 2024-12-27 by schoolbust