Brandawg93/PeaNUT #132: [Bug]: POST commands not working on Tecnoware ERA PLUS 1100

nuzzzen, 2024-12-25:
### Describe the bug

No POST endpoint is working on my UPS after setting up everything. GET endpoints work and return consistent data.

For example, this is what `devices/[ups]/commands` endpoint returns:

```json
[
  "beeper.toggle",
  "load.off",
  "load.on",
  "shutdown.return",
  "shutdown.stayoff",
  "shutdown.stop",
  "test.battery.start.quick"
]
```

 I would like to at least, execute commands like `shutdown.return` or similar.
 I already tested using the `upscmd` command and found no issues.
 
 Also, calling `devices/[ups]/rwvars` I'm getting no writable vars: '[]'

### Debug Output

500 "Failed to save variable beeper.toggle on device nutdev1"

```
at s.getCommand (.next/server/chunks/250.js:2:1072)
    at async s.setVar (.next/server/chunks/250.js:3:2198)
    at async p (.next/server/app/api/v1/devices/[device]/var/[param]/route.js:1:1849)
Error: Invalid response: ERR VAR-NOT-SUPPORTED
```

### Steps to reproduce

1. Setup nut and peanut
2. POST `/devices/[ups]/var/beeper.toggle` with any kind of json body (i.e `{}`)
3. See failed response

### Device Type

Tecnoware ERA PLUS 1100

### UPS Device

Tecnoware ERA PLUS 1100 - driver nutdrv_qx

### Last Working Version

N.A.

Brandawg93, 2024-12-26:
I would expect that the `POST` to save variables would not work if your ups doesn't have any writable variables. But ironically, I just started working on the new `POST` api call to execute commands. #127 I think the latest test version even has it included. You can test it via `brandawg93/peanut:test`. Let me know if it works for you!

Brandawg93, 2024-12-26:
Commands via `POST` is now available in v4.3.0! Let me know if it works for you.

closed 2024-12-26 by Brandawg93

nuzzzen, 2024-12-26:
Hell yeah faster than light haha! 

I confirm everything works as expected now. I suppose you could add my device in the list of the supported devices, since everything works as it should.

- Brand: `Tecnoware`
- Model: `ERA PLUS 1100`
- Driver: `nutdrv_qx`

Thank you very much.

Brandawg93, 2024-12-27:
Awesome! Will do.