Brandawg93/PeaNUT #136: [Feature Request]: notification integrations

rmlamarche, 2024-12-31:
### Enhancement of issue

The ability for PeaNUT to send notifications (via things like smtp, ntfy, webhooks, etc.) when certain conditions are met

**Example use-cases:**
- UPS status changes from on-line to on-battery
- utilization above a certain threshold
- battery time remaining below a certain threshold
- input/output voltage above/below a certain threshold
- nut server becomes unreachable

### Solution

Configurable notification providers for popular notification tools (smtp, ntfy, generic webhooks, ...). Monitoring period could possibly use the same timer that the InfluxDB exporter uses. Some notification types may require PeaNUT to "memorize" the last known state of each UPS (eg. status changing from OL to OB), but this memory doesn't necessarily need to be persisted to a database, it could certainly just be comparing to the previous reading in-memory.

Presumably these settings could be configured from the PeaNUT GUI and would be persisted in the settings YAML file.

### Alternatives

I've considered using changedetection.io for this, and pointing it at the PeaNUT web client. This would certainly work, but I still think this would be great to have natively in PeaNUT, especially since it would fit in nicely with the periodic monitoring that PeaNUT server is already doing for InfluxDB exports.

I think home assistant also has some UPS monitoring add-ons that likely can be used with their automations and mobile app notifications, I've not tried this but I suppose it is an alternative.

I would prefer to only use one tool to monitor my UPS status rather than many, and I really like using PeaNUT.

### Additional Context

If this idea fits in with your vision for PeaNUT, I'm happy to help come up with more example use-cases, integration ideas, or even work on it/contribute to it. If this is considered out of scope for PeaNUT, I understand but wanted to throw the idea out there!

rmlamarche, 2025-01-11:
I've started to experiment with this locally. Below is how I envision it working (example below using a self-hosted ntfy instance).

`NOTIFICATION_INTERVAL` - frequency at which PeaNUT polls the NUT server
`NOTIFICATION_PROVIDERS` - configure various notification providers, each with their own "config" (which would be specific to that notification provider, e.g. ntfy has tags/priorities/etc that can be set), and "triggers" (described below)

The triggers I've experimented with so far:
- `variable` **changes** - any change from the previous state/previous reading will result in a notification
- `variable` **is_above** `targetValue` - if the value exceeds the threshold, a notification is triggered
- `variable` **is_below** `targetValue` - if the value drops below the threshold, a notification is triggered

```yaml
NUT_SERVERS:
  - HOST: 192.168.1.1
    PORT: 3494
    USERNAME: ''
    PASSWORD: ''
INFLUX_HOST: ''
INFLUX_TOKEN: ''
INFLUX_ORG: ''
INFLUX_BUCKET: ''
INFLUX_INTERVAL: 10
NOTIFICATION_INTERVAL: 10
NOTIFICATION_PROVIDERS:
  - name: 'ntfy'
    triggers:
      - variable: input.voltage
        operation: changes
      - variable: battery.runtime
        operation: is_below
        targetValue: 1300
    config:
      server_url: https://ntfy.example.com/
      topic: my-ntfy-topic
      priority: 1
      tags: []
```

[image]

I have not done any work in the UI yet or worked in any tests or documentation, I'm still just experimenting, but if there's interest I'm willing to keep working on it/polish it and submit a PR

Brandawg93, 2025-01-11:
This is AWESOME! Notifications and persistent data are my two top features I want to implement. Honestly, my dream is to have the project look and feel similar to [beszel](https://github.com/henrygd/beszel). Thanks so much for your work on this! Definitely keep it up. There's lots of interest. Feel free to submit a PR and I can review and assist with code there. Also take a look at how notifications are set up in the [NUT server itself](https://networkupstools.org/docs/user-manual.chunked/Advanced_usage_scheduling_notes.html).

rmlamarche, 2025-01-12:
Glad you like the direction so far! I cleaned up what I had in progress and opened a draft PR #151 

Definitely more work to do with documentation/test suites & I'm open to feedback, particularly with the UI. I'm also unsure how the translations work, are those automatic?

Brandawg93, 2025-01-12:
They used to be but my trial period ended for Crowdin... But there is a really awesome extension for VS Code called [i18n Ally](https://github.com/lokalise/i18n-ally) that makes localization so much easier.

vladaurosh, 2025-01-13:
I was about to mention what @Brandawg93 already mentioned, NUT itself provides a way to do anything when state changes.
For example, I have in /etc/nut/upssched.conf:
```
CMDSCRIPT /etc/nut/upssched-cmd
...
AT ONBATT * EXECUTE onbatt 
```
Then in /etc/nut/upssched-cmd I have something like:
```
#! /bin/bash

function pushover {
    curl -s -F "token=***************************" \
    -F "user=**************************************" \
    -F "title=$1" \
    -F "message=$2" https://api.pushover.net/1/messages.json
}

case $1 in
       onbatt)
          title="UPS running on battery"
          message="UPS started to run on battery"
          logger -t upssched-cmd "$title"
          # push message to pushover
          pushover "$title" "$message"
          ;;
       onbatt_shutdown)
....
```

In this example I use pushover.

Not that I am against option to send notifications from PeaNUT but ultimately, since NUT server executes various actions when 
ups state changes, you'd want to manage everything from NUT.

rmlamarche, 2025-01-14:
Yeah, I’m aware of the ability to configure actions in NUT when events occur, but I don’t think there’s any downside to offering this functionality in PeaNUT. There are even cases where (in my opinion) handling notifications from PeaNUT may be preferred over the native solution.

For example, one of my NUT servers is the OPNsense NUT plugin, which does not provide a mechanism for configuring the nut .conf files. Sure I could open a terminal and go mess with the config, but I prefer not to manually edit config files that could be overridden in an upgrade, and my changes also wouldn’t be included in my OPNsense backup/restore process.

It would offer:
1. A mechanism for notifications when the advanced NUT config is obscured/abstracted away from the user (e.g. OPNsense NUT plugin, but it’s likely there are other examples of this out in the wild)
2. Notifications configured in one, centralized tool for many NUT servers
3. Ease of use for people who don’t want to fiddle with NUT config
4. More options for ways to “trigger” such notifications (not limited to what NUT provides). AFAIK (I could be wrong) there’s no way for NUT to natively send you a notification if, for example, ups load exceeds 80% capacity, or battery runtime is less than 15 minutes, etc.
5. No downside to people who prefer to use the native NUT server events; it would just be an optional alternative/additional tool.

yshah-aromatech, 2025-01-30:
This is very exciting. currently the service going to be added is ntfy? 

Big thanks to everyone working on this.

Brandawg93, 2025-02-09:
I'm still working on this. But in the meantime, v5.2.0 now has prometheus support for alerts! 

The endpoint is `/api/v1/metrics`. And examples are available.