Brandawg93/PeaNUT #139: [Bug]: UI kinda "broken" in 4.4.0

vladaurosh, 2025-01-03:
### Describe the bug

Well, not broken, but some things are bad.

I don't see values for input/output voltage when I position the cursor on dots.
Also, clicking on Input or Output boxes doesn't work anymore.
[image]

Same thing here, but even worse as range is 0-100-200-300-400 so it looks like it's close to 0. 
[image]

Also, power should be in W (watts) not V (volts).

Another thing, "Nominal Power" label is badly position, it's hitting "Real Power" label.

### Debug Output

N/A

### Steps to reproduce

 N/A

### Device Type

rpi 5

### UPS Device

Eaton 5P

Brandawg93, 2025-01-04:
Thanks for the report! This definitely makes sense as I've recently switched from [Chart.js](https://www.chartjs.org) to [MUI X](https://mui.com/x/react-charts/). I'll look into making the charts look more similar to the old version.

Could you send me a screenshot of what the old charts looked like for you? I'm curious to see what the nominal lines looked like for you considering the your real power and nominal real power are so different.

vladaurosh, 2025-01-05:
Thanks for quick reply,
Here are the screenshots of how it looked like for me in 4.3.0.
[image]
[image]
[image]

Brandawg93, 2025-01-05:
Interesting. So you don't see the nominal line in certain cases. Ok I just published a new test version that should fix everything with the latest changes. `brandawg93/peanut:test`. Let me know if that fixes this for you.

vladaurosh, 2025-01-05:
Thanks.

It does look a bit better. But, there are still couple of issues:
[image]
Line colors for input and output voltage are almost the same, maybe change one to something more distinct?

[image]
If input and output voltage are same, box will just show output voltage. 
Input and Output checkboxes are still not clickable.
Nominal input voltage label too close to Input and Output labels and checkboxes, maybe move them up a bit?

[image]
Wrong value for Nominal real power, showing the same as real power, should be 420W in my case.
Also, still showing V on the Y axis.

Brandawg93, 2025-01-05:
Awesome writeup! This really helps. Let me take a look at these and I'll get back to you.

Brandawg93, 2025-01-06:
Could you try the latest test version? Should be much better now.

vladaurosh, 2025-01-06:
Hi @Brandawg93  Just tried latest test version, I'd say it's almost there, just couple of issues left.
Checkboxes are clickable, that's good.

[image]
[image]

Two issues here:
[image]
When both input and output voltage boxes are checked, graph disappears. :)

[image]

Second issue, when input and output voltages are same, hovering over dot shows Output voltage (with old behavior it was showing both).

[image]

Nominal realpower stil showing same as real power. 

Something is wrong here, at some point Nominal real power just disappeared. :| 
[image]

Y axis is fixed, showing W now. :)

Brandawg93, 2025-01-09:
1. The graph disappearing is expected considering the only two datasets are deselected.
2. The tooltip hovering issue is due to how the charts are created via MUI X. For now, you should be able to deselect a dataset and see the tooltip.
3. What is your expected real power and nominal real power? Everything I've looked at in the code points to the correct procedure, but it could be a MUI X issue still.
4. The nominal real power line may disappear at times because it is only a reference line.

Unfortunately, the new chart package is still pretty new but the old one just had so many dependencies and build issues. I'm still looking into getting the new charts to be the same as the old ones, but I may have to wait for the new charts maintainers to implement some features. For now, I'm going to release these couple of fixes in a new version.

Brandawg93, 2025-01-09:
I did actually find some discrepancies between the voltage chart and watts chart. I've created a new test build that may address those issues.

vladaurosh, 2025-01-09:
Hi @Brandawg93 

Just tried latest test version. There are some issues:
- nominal input voltage somehow disappeared, when both Input and Output voltage boxes are unchecked. Could be due to Y axis scaling (230V should be nominal voltage)
[image]
- when one box is checked it appears again
[image]
- looks like in latest test Power diagram became Input voltage somehow, showing power values (at least digits are what I'd expect for power), and no nominal real power is shown
[image]

To answer your question about real power of my UPS, it's 420W.

Another weird thing is Y axis scaling in voltage diagram. It shows +- several volts when both Input and Output voltages are shown, but when just one is shown it shows 0 to 250V. A little bit of inconsistence there, if you ask me. :)

Brandawg93, 2025-01-10:
Too much copy/pasta on my end. 😄 I just pushed a new test version that should fix most of this.

vladaurosh, 2025-01-10:
Thanks @Brandawg93 
Just tried latest test version. Looks better but still couple of issues:
1. with both Input and Output boxes unchecked it looks ok.
[image]
2. but if one box is checked 
3.

vladaurosh, 2025-01-10:
Thanks @Brandawg93 
Just tried latest test version. Looks better but still couple of issues:
1. with both Input and Output boxes unchecked it looks ok.

[image]

2. but if one box is checked Nominal voltage tooltip is again very close to Input and Output tooltips. Since there's a lot of free space in Output/Input Voltage card, maybe move them up a bit? Also blue line is a bit over "Nominal Input Voltage" label. 
Also, in previous screenshot voltage range was 230-235V while on second one is 0-250V?
[image]

3. Label issue in Power diagramas well, maybe move tooltip a bit more up since there's free space in the card?
[image]

Brandawg93, 2025-01-11:
Ok. I fixed the padding in the latest test build. It should be more responsive now.

vladaurosh, 2025-01-11:
Thanks. 
Looks better now. 

Any idea about the change of Y axis scaling? Is that related to MUI X?

Brandawg93, 2025-01-11:
It is. I've been playing with the [domain limit](https://mui.com/x/api/charts/axis-config/#axis-config-prop-domainLimit) a lot to see if I can make it more similar to the old charts, but for some reason, it really wants the min to be 0 when there is only one dataset.

Brandawg93, 2025-01-13:
The last touch ups have now been published in v4.4.3. Thanks for all the debugging and testing. If you find any other issues, just let me know.

closed 2025-01-13 by Brandawg93