Brandawg93/PeaNUT #142: [Bug]: chokidar file watching does not work with nfs mounts

rmlamarche, 2025-01-05:
### Describe the bug

chokidar file watching constantly thinks there are changes to my settings.yml file, which triggers the addOrUpdateJob event to infinitely run. This is causing increased load on my NUT server & InfluxDB, and has made the PeaNUT GUI _very_ sluggish when refreshing data.

### Debug Output

I added some extra console logging to debug this, chokidar onchange event handler is running multiple times per second, and as a result the addOrUpdateJob is constantly getting run

[image]

### Steps to reproduce

1. Run PeaNUT with settings.yml mounted via NFS

### Device Type

on-prem kubernetes cluster

### UPS Device

APC ES 600M1

### Last Working Version

4.3.0

Brandawg93, 2025-01-05:
I'm looking into this now. Just leaving this thread here for future reference:

https://github.com/paulmillr/chokidar/issues/201

Brandawg93, 2025-01-05:
It looks like you can set the environment variable `CHOKIDAR_USEPOLLING` to true to force chokidar to poll for NFS. Does that work for you?

Brandawg93, 2025-01-06:
Let me know if you still have issues with this.

rmlamarche, 2025-01-06:
I have tried enabling polling and tweaking the polling interval. The problem (I believe) is that the modification timestamp on the file is getting "bumped" every time a YamlSettings object is instantiated. If I continuously stat the file, I can see the change + mod times continue to be updated infinitely.

I think that's happening is the YamlSettings constructor calls fs.writeFileSync, and every influxdb task instantiates a YamlSettings object, and hence the infinite loop.

I believe there is an issue here, and it might not be unique to my kubernetes or NFS setup. If there's anything more I can do to help debug let me know.

rmlamarche, 2025-01-06:
I built a custom image with all the chokidar stuff disabled, and I still see the mod time on the settings.yml file changing every time the influxdb task runs. I believe there is an infinite loop of:
1. influxdb task runs
2. new YamlSettings object created
3. YamlSettings.save() flushes file contents to disk
4. chokidar detects change
5. influxdb task runs immediately as a result of chokidar event

Brandawg93, 2025-01-07:
Take a look at #143 and let me know if that fixes the issue for you.

rmlamarche, 2025-01-07:
Just tested with those changes, I think it's a step in the right direction but doesn't solve the whole problem. There is still an infinite loop when the influx writer task runs, now it's just an infinite loop every 10 seconds instead of multiple times per second. But there are also a lot of other instances of YamlSettings objects being created -- each one will save the .yaml file even if there are no changes to be saved.

I think the YamlSettings class constructor calling save() is the root cause of the problem. Every time a YamlSettings object is created (which happens quite often in the code base) it will trigger the chokidar change event. I think the YamlSettings behavior needs to be more intelligent about when to flush to disk to play nicely with chokidar.

Brandawg93, 2025-01-08:
Definitely makes sense. Check out the latest commit and see if that helps any. Now it only saves if the yaml file doesn't exist yet.

ae09fdb48c0e065c2d2783c810fafd6f87bf5dfa

rmlamarche, 2025-01-08:
This definitely fixed the infinite write-read chokidar loop!

With the YamlSettings save changes, I think the chokidar watcher change could probably be reverted? I actually can't get chokidar working at all now, it doesn't seem to be triggering when the settings change (with or without CHOKIDAR_USEPOLLING enabled/disabled). I can do more testing later this week on this... but right now the only way I can modify my InfluxDB settings is to restart peanut.

I'll post an update here after I can do some more thorough testing/debugging. Thanks for the work you've put into this so far!

Brandawg93, 2025-01-09:
I've released this in v4.4.1. If you find anymore issues in your testing, feel free to reopen this or comment. Thanks for helping test this!

rmlamarche, 2025-01-10:
v4.4.1 is working flawlessly with chokidar on NFS. Thanks again!