Brandawg93/PeaNUT #145: [Feature Request]: Provide better upfront configuration example

basher83, 2025-01-10:
### Enhancement of issue

I was having issues connecting the InfluxDB as I wasn't sure exactly how the parameters were suppose to look. For example I didn't know if for the bucket I needed the bucket name or the alphanumeric identifier for the bucket. I had to go search through the issues and finally found issue #131 which referred back to https://github.com/Brandawg93/PeaNUT/tree/main/examples/persistant-data/docker-compose.yml which finally gave me my answer. It was noted in the wiki https://github.com/Brandawg93/PeaNUT/wiki/Simple-Default-Configuration and the bottom under settings.yml but again this only shows the fields and not what is expected to be in them.

### Solution

Change https://github.com/Brandawg93/PeaNUT/wiki/Simple-Default-Configuration and at the bottom under settings.yml

NUT_SERVERS:
  - HOST: nut-upsd 
    PORT: 3493
    USERNAME: upsmon
    PASSWORD: supersecretpassword1234
INFLUX_HOST=http://influxdb:8086
INFLUX_TOKEN=super_secret_token
INFLUX_ORG=home
INFLUX_BUCKET=ups
INFLUX_INTERVAL=3

### Alternatives

Add the below to the main README

NUT_SERVERS:
  - HOST: nut-upsd 
    PORT: 3493
    USERNAME: upsmon
    PASSWORD: supersecretpassword1234
INFLUX_HOST=http://influxdb:8086
INFLUX_TOKEN=super_secret_token
INFLUX_ORG=home
INFLUX_BUCKET=ups
INFLUX_INTERVAL=3

Brandawg93, 2025-01-10:
Thanks for the heads up! I've added it to the simple configuration wiki. I think I'll leave the YAML configuration wiki alone for now, since those values could be different for different users.

closed 2025-01-10 by Brandawg93

basher83, 2025-01-11:
Very true, love the project btw. I'm roughly one year into homelabing and have fallen in love with the open source community projects. I'm happy to provide any insight to help you make thigs easier and better for the rest of us!

Brandawg93, 2025-01-11:
Thanks so much! I've been homelabbing for a while now and just enjoy making these little projects in my spare time. If you haven't already come across it, check out [selfh.st](https://selfh.st). It's my go-to for new projects and updates.

And feel free to provide feedback or support for peanut wherever you see fit. That's what is so awesome about the open source community! We all help each other out. 🦾

basher83, 2025-01-11:
Thanks for the site referral! Always great to find those cool projects no one knows about yet