Brandawg93/PeaNUT #149: Crowdin project is not visible

ildoc, 2025-01-11:
### Describe the bug

I'd like to contribute to the translations, but the crowdin project has limited visibility

### Debug Output

.

### Steps to reproduce

1. go to https://crowdin.com/project/nut-dashboard
2. Something went wrong
Project does not exist or access to this project is limited.

### Device Type

.

### UPS Device

.

Brandawg93, 2025-01-11:
Yep. Unfortunately, Crowdin is only free for a certain amount of time and my trial ran out. But feel free to contribute directly on the GitHub! It's relatively straightforward. Just follow the same flow as what you see in the [locales folder](https://github.com/Brandawg93/PeaNUT/tree/main/src/client/i18n/locales). If you run into any issues or have any questions, just let me know.

ildoc, 2025-01-11:
understood, thanks!
I opened  #150, only minor corrections, but they were bugging me :smile:

Brandawg93, 2025-01-11:
Awesome! I'll take a look at them there. Thanks again.

closed 2025-01-11 by Brandawg93