Brandawg93/PeaNUT #154: [Bug]: State shown just when UPS is online

vladaurosh, 2025-01-13:
### Describe the bug

When UPS is online we see this in UI.
[image]

But when UPS is in for example, Online Charging state nothing is shown.
[image]
[image]

### Debug Output

N/A

### Steps to reproduce

N/A

### Device Type

rpi 5

### UPS Device

Eaton 5P

vladaurosh, 2025-01-18:
Thanks. Will need to try with dummy ups. :) 
I though that status was pretty common, when ups is back online but charging.

vladaurosh, 2025-01-18:
Hey @Brandawg93 
Just tried with latest dev and got this

[image]

Brandawg93, 2025-01-18:
What was your UPS status there? It may be something else I don't have in the list.

Brandawg93, 2025-01-18:
I just pushed a new test that should just fallback to the actual status in the grid.

vladaurosh, 2025-01-18:
Status is OL CHRG. 
I'll try with new test image, thanks.

vladaurosh, 2025-01-18:
Looks good now, thanks. 

[image]

Brandawg93, 2025-01-19:
Awesome! Fixed in v4.4.4. Thanks for the help debugging.