Brandawg93/PeaNUT #159: [Feature Request]:  Display Watt Usage in Addition to UPS Load

indhayarf, 2025-01-28:
### Enhancement of issue

Currently, the web UI displays the UPS load as a percentage (e.g., ups.load). While this is useful, it does not provide users with a clear understanding of the actual power consumption in watts. I propose adding a feature to calculate and display the watt usage of the UPS alongside the existing load percentage. This enhancement would improve the usability and clarity of the monitoring interface.

### Solution

Solution
The watt usage can be calculated using the following formula:

Watts = (UPS Load (%) / 100) * UPS Rated Power (in Watts)
For example, if the UPS is rated at 1200W and the load is 50%, the watt usage would be:

Watts = (50 / 100) * 1200 = 600W

Implementation Steps:

1. Add a new field in the web UI to display watt usage (e.g., "Watt Usage: 600W").

2. Dynamically calculate watt usage based on the ups.load value and the UPS's rated power (1200W for a 2000VA/1200W UPS).

3. Ensure the watt usage updates in real-time alongside the load percentage.

### Additional Context

Why This Matters: Displaying watt usage provides users with a more intuitive understanding of their UPS's power consumption, helping them monitor how close they are to the UPS's maximum capacity (1200W in this case).

Configurability: The rated power of the UPS (1200W) could be made configurable in the settings to support different UPS models.

User Experience: The watt usage could be rounded to 2 decimal places for readability and displayed near the existing load percentage for consistency.

Brandawg93, 2025-01-28:
Try clicking on the gauge charts. Is that what you’re referring to?

indhayarf, 2025-01-29:
nothing happens when I click the gauge charts, no popup or anything

Brandawg93, 2025-01-29:
That's odd. When you click the gauge charts, it should show the wattage info. What browser/device are you using?

paulbaird87, 2025-02-04:
@Brandawg93
Just an idea! Would love to have watts in place of the "Battery Runtime" in big tiles at the top. You can slide the battery runtime data into the battery charge field or have the number just under the "Online" in the top right?

I do agree though, many of us think and speak in Watts especially those of us unfortunate enough to live in countries where electricity is overpriced.

indhayarf, 2025-02-04:
i am running PeaNUT in a docker container on TrueNAS scale 24.10, and i tried accessing it using both mozilla and chrome, same result the gauge charts seem to be static, nothing if you hover above them or even if you click on them? do you have of snapshot of the expected behavior? 

and as mentioned by @Brandawg93 the main reason i need Watts is to monitor power usage because of overpriced electricity

Brandawg93, 2025-02-04:
I’ll have to check if there is a bug or something. The expected behavior is that you can click the gauge charts and see calculated wattage. Not sure why it isn’t happening for some. 

Here is the previous issue. #75

Brandawg93, 2025-02-09:
Try the latest version (v5.2.0) and let me know if it helps. I added some accessibility tags to the gauges' HTML to hopefully tell the browser it's clickable.

paulbaird87, 2025-02-12:
Can confirm this works and shows wattage to one decimal place. Thanks @Brandawg93  
May I also ask were you involved in benphelps/homepage integration? Would love to see Watts in the "Allowed Fields" for that widget.

Brandawg93, 2025-02-12:
@paulbaird87 Yes, I added support for PeaNUT there. I currently have the api set up to only display info coming from your device, and I'd prefer not to introduce calculated fields there. Maybe homepage allows calculations with the custom widgets?

https://github.com/Brandawg93/PeaNUT/blob/main/src/client/components/wrapper.tsx#L175

Brandawg93, 2025-02-12:
The fix is now live in v5.2.1.

@paulbaird87 Feel free to continue discussion here or start a new Discussions thread. The homepage widget is a hot topic, so that may be a good thread to start.

closed 2025-02-12 by Brandawg93

rufftruffles, 2025-09-27:
Running the latest v5.15.0 docker and I still don't see wattage anywhere? Only volts on the graph.