Brandawg93/PeaNUT #164: [Bug]: Error in browser when accessing my APC

NightHawkATL, 2025-01-31:
### Describe the bug

I am getting an error when trying to access the data from my APC unit. This was working before 5.0.

[image]

[image]

### Debug Output

[image]

### Steps to reproduce

1. go to web UI
2. select an APC UPS from the list.
3. See error.

### Device Type

Linux 22.04 VM running on Proxmox with USB passthrough

### UPS Device

Cyberpower 1500VA x2 (LX1500GU), APC 1300VA (XS 1330G)

### Last Working Version

v4.4.4

Brandawg93, 2025-01-31:
Yep. Definitely an issue. v5.0.1 coming up soon! 😅

Brandawg93, 2025-01-31:
Ok. v5.0.1 should be live. Please let me know if this fixes your issue.

closed 2025-01-31 by Brandawg93

NightHawkATL, 2025-01-31:
Fixed! Thanks!