Brandawg93/PeaNUT #166: [Bug]: Latest 5.0.1 not working

vladaurosh, 2025-02-01:
### Describe the bug

Getting this when updated from 4.4.4 to 5.0.1
Tried different browsers. clearing cache, but issue persists.

[image]

Container itself is healthy:
```
> peanut@5.0.1 start
> HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js

   ▲ Next.js 15.1.6
   - Local:        http://localhost:8080
   - Network:      http://0.0.0.0:8080

 ✓ Starting...
 ✓ Ready in 232ms
```

### Debug Output

Error: Minified React error #185; visit https://react.dev/errors/185 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.

### Steps to reproduce

1. 

### Device Type

Rpi 5

### UPS Device

Eaton 5P 650

vladaurosh, 2025-02-01:
Same error in 5.0.0

Brandawg93, 2025-02-01:
Does this happen when you load the page? Or did you click on a certain button?

vladaurosh, 2025-02-01:
Can't get the page to open, doesn't load at all.
/api/v1/devices returns expected.

vladaurosh, 2025-02-01:
[image]

Getting this and then just error message I've posted above.

Brandawg93, 2025-02-01:
Got it. And just to confirm, your config didn’t get corrupted during the upgrade?

vladaurosh, 2025-02-01:
Current content of config/settings.yml:
```
NUT_SERVERS:
  - HOST: <nut_host>
    PORT: 3493
INFLUX_HOST: ''
INFLUX_TOKEN: ''
INFLUX_ORG: ''
INFLUX_BUCKET: ''
INFLUX_INTERVAL: 10
WEB_HOST: 0.0.0.0
WEB_PORT: 8080
NUT_HOST: <nut_host>
NUT_PORT: 3493
```

I guess if it was corrupted, /api/v1/devices wouldn't work?

Brandawg93, 2025-02-01:
Yep. Just making sure. Could you post the output of the devices api call either here or email me at brandawg93@proton.me? Whatever you’re more comfortable with. That way I can mock your system.

vladaurosh, 2025-02-01:
Sure. Here it is.
`[{"battery.capacity":9,"battery.charge":100,"battery.charge.low":"20","battery.charge.restart":"0","battery.charger.status":"resting","battery.energysave":"no","battery.energysave.delay":"300","battery.energysave.load":"5","battery.protection":"yes","battery.runtime":932,"battery.type":"PbAc","battery.voltage":13.1,"battery.voltage.nominal":12,"device.mfr":"EATON","device.model":"5P 650","device.serial":"<serial_number>","device.type":"ups","driver.name":"usbhid-ups","driver.parameter.pollfreq":30,"driver.parameter.pollinterval":1,"driver.parameter.port":"auto","driver.parameter.productid":"ffff","driver.parameter.serial":"<serial_number>","driver.parameter.synchronous":"no","driver.parameter.vendorid":463,"driver.version":"2.7.4","driver.version.data":"MGE HID 1.40","driver.version.internal":0.41,"input.current":0.8,"input.frequency":49.9,"input.frequency.extended":"no","input.frequency.nominal":50,"input.sensitivity":"normal","input.transfer.boost.low":"184","input.transfer.high":"294","input.transfer.low":"160","input.transfer.trim.high":"265","input.voltage":236.2,"input.voltage.extended":"no","input.voltage.nominal":230,"outlet.1.autoswitch.charge.low":"0","outlet.1.delay.shutdown":"65535","outlet.1.delay.start":"3","outlet.1.desc":"PowerShare Outlet 1","outlet.1.id":1,"outlet.1.status":"on","outlet.1.switchable":"yes","outlet.2.autoswitch.charge.low":"0","outlet.2.delay.shutdown":"65535","outlet.2.delay.start":"6","outlet.2.desc":"PowerShare Outlet 2","outlet.2.id":2,"outlet.2.status":"on","outlet.2.switchable":"yes","outlet.desc":"Main Outlet","outlet.id":"0","outlet.switchable":"no","output.current":0.7,"output.frequency":49.9,"output.frequency.nominal":50,"output.powerfactor":0.76,"output.voltage":236.2,"output.voltage.nominal":"230","ups.beeper.status":"enabled","ups.delay.shutdown":"20","ups.delay.start":"30","ups.efficiency":88,"ups.firmware":"03.18.0035","ups.load":26,"ups.load.high":"105","ups.mfr":"EATON","ups.model":"5P 650","ups.power":165,"ups.power.nominal":650,"ups.productid":"ffff","ups.realpower":126,"ups.realpower.nominal":420,"ups.serial":"<serial_number>","ups.shutdown":"enabled","ups.start.auto":"yes","ups.start.battery":"yes","ups.start.reboot":"yes","ups.status":"OL","ups.test.interval":"2592000","ups.test.result":"Done and passed","ups.timer.shutdown":-1,"ups.timer.start":-1,"ups.type":"offline / line interactive","ups.vendorid":463}]`

Brandawg93, 2025-02-01:
Thanks! Yep I see the same problem too. It'll take me some time to debug. I'll message back here when I have a fix.

Brandawg93, 2025-02-01:
Found the fix. Try out the latest test version and let me know if it fixes is for you. `brandawg93/peanut:test`.

Also, on a different note, your UPS is extremely thorough. I'm seeing properties I've never seen before. Neat! Would you mind sharing the output of `api/v1/devices/[device]/commands`? That way I could add more commands to the app.

vladaurosh, 2025-02-01:
Yeah, seems to be working now. Thanks. 
Eaton is more enterprise grade UPS. 
Here are the commands:
```
[
  "beeper.disable",
  "beeper.enable",
  "beeper.mute",
  "beeper.off",
  "beeper.on",
  "load.off",
  "load.off.delay",
  "load.on",
  "load.on.delay",
  "outlet.1.load.off",
  "outlet.1.load.on",
  "outlet.2.load.off",
  "outlet.2.load.on",
  "shutdown.return",
  "shutdown.stayoff",
  "shutdown.stop",
  "test.battery.start.deep",
  "test.battery.start.quick",
  "test.battery.stop"
]
```

BTW, I really like new design. :) Couple of visual bugs I've noticed:

Nominal voltage somehow disappears:
[image]

[image]

Maybe move voltage more to the right, number is very close to Output Voltage label. 
[image]

Brandawg93, 2025-02-02:
I just published a new test version that should fix those issues and add a new graph for your device. Let me know if that looks better to you! `brandawg93/peanut:test`

vladaurosh, 2025-02-02:
Thanks @Brandawg93 
That looks really good now. I see new graph Power (in VA) has appeared now. Is this going to be included by default in new version? Any chance to make that one as optional?

Brandawg93, 2025-02-02:
Hm. That's a good idea. Try the latest test. I've made the charts collapsible.

vladaurosh, 2025-02-02:
That's very nice update @Brandawg93 . :)

Brandawg93, 2025-02-03:
Awesome. This is now live in v5.1.0. Thanks for all the help testing!

closed 2025-02-03 by Brandawg93