Brandawg93/PeaNUT #171: [Feature Request]: Make Key/Value table collapsible

vladaurosh, 2025-02-09:
### Enhancement of issue

Hey @Brandawg93 
Is it possible to make Key/Value table collapsible (same as charts)?

### Solution

N/A

Brandawg93, 2025-02-09:
Good idea. I have a new test version with this added. Let me know if it works for you.

`brandawg93/peanut:test`

vladaurosh, 2025-02-10:
Thanks. This looks good, though maybe it should have some label, similar to the charts?

[image]

Brandawg93, 2025-02-10:
Good point. I just made a new test version that's a bit cleaner.

vladaurosh, 2025-02-10:
Cheers. That looks much better now.

Brandawg93, 2025-02-12:
This is now live in v5.2.1.

closed 2025-02-12 by Brandawg93