Brandawg93/PeaNUT #172: [Doc]: Homepage custom API for runtime change from text to duration

ForsakenRei, 2025-02-09:
### Enhancement of issue

Current the custom API example uses
```    
- field: battery.runtime
   label: Battery Runtime
   format: text
```
with `format: text` homepage shows UPS runtime in seconds, and it's quite hard to read. I forgot in which update but custom API can have a duration format to make time a little more user friendly. https://gethomepage.dev/widgets/services/customapi/

### Solution

Update the doc with
```    
- field: battery.runtime
   label: Battery Runtime
   format: duration
```
Here's what it looks like on my homepage

[image]

Brandawg93, 2025-02-09:
Cool! I didn't know that. I've updated the docs. Thanks for the heads up!

closed 2025-02-09 by Brandawg93

ForsakenRei, 2025-02-09:
> Cool! I didn't know that. I've updated the docs. Thanks for the heads up!

NP, and I really like this small utility!