Brandawg93/PeaNUT #174: [Bug]: UPS settings deleted if communication with UPS fails

mx5gr, 2025-02-10:
### Describe the bug

I entered two UPSs within my network to the settings of PeaNUT, both of them displayed perfectly within the PeaNUT UI.

The next morning, one of the two Raspis that monitor the UPSs (one each) was off. I noticed that, upon loading, the PeaNUT interface took a while to load (probably due to NUT server connect timeout expiration) and, afterwards, we were presented with an "Internal Error" message, with a logout button underneath.

As soon as we relaunched PeaNUT, we were asked to enter the NUT server information from scratch. We noticed that the settings.xml file had been completely purged of information.

The above behavior was replicated in another 2 PeaNUT installations.

The application should try to connect to already declared NUT servers and, in the case of a connect failure, to mark this server as disconnected within the app, with the chance to reconnect to it (e.g. via a new button or through the existing connect button). Surely, there should be no purging of the entire configuration file, including any references to other NUT servers and/or influx instance.

### Debug Output

Nothing recorded to the docker log

### Steps to reproduce

1. Add a NUT server to PeaNUT
2. Turn off NUT server
3. Open PeaNUT UI

### Device Type

Raspi 4

### UPS Device

Eaton 1200

mx5gr, 2025-02-10:
After further investigation, the issue lies with the inability of the UI to handle the connection timeouts to an already declared NUT server.

Instead of skipping the failed connection (to the NUT server) and try the next one, even if only 1 connection of the list of NUT servers fails, the UI hangs, showing "Internal server error". The actual "logout" button that is displayed underneath does the file purging, which should never occur! The settings should remain within the file and, upon docker container relaunch, the script should retrieve the settings from there, instead of manually inputting them again every time! Thus, the logout button should be clearly labelled as "Settings Purge" and - possibly - another button to be added which actually restarts the script, with the existing settings.

In any case, the handling of a failed connection to a NUT server should be handled in a better way, as the UI becomes unusable if even only 1 NUT server in the list is not connectable.

Brandawg93, 2025-02-10:
Good point. That page was created before the multi server update and needs a refresh. I'll keep you posted when I have something to test.

Brandawg93, 2025-02-14:
This should be fixed in v5.3.0. If you have any other issues with it, just let me know.

closed 2025-02-14 by Brandawg93