Brandawg93/PeaNUT #178: [Feature Request]: Womepage Widget to display Watts

paulbaird87, 2025-02-12:
### Enhancement of issue

Would love to see watts in the Homepage widget if possible.....More than happy to request this with @benhelps.U

### Solution

Unsure if Peanut can pass though calculated values or if that needs to be done in the Peanut Homepage widget itself?

Brandawg93, 2025-02-21:
@shamoon Do you know if this can be done with the custom API widget?

shamoon, 2025-02-21:
Anything that gets spit out by the peanut API can be used in a customapi widget

shamoon, 2025-02-21:
scale is applied per field, e.g.

```
- field: key5
  label: Power
  format: float
  scale: 0.001 # can be number or string e.g. 1/16
  suffix: "kW"
```

(from the docs)

iAmZyk, 2025-09-17:
Keen for this one. Would be a great add.