Brandawg93/PeaNUT #179: Info: more info about what is needed for what

aroundmyroom, 2025-02-12:
When I configured PeaNUT I could access my UPS without username and password, although I had configured a username and password in NUT
So as there was no reason for me to enter a username / password I did not use it 

until today I want to see if the test options did function. Although the system reports it is running a test, it does not do anything
cause I did not enter a username / password. So my test (ie. see if the beeper was enabled or disabled) was not changing anything in the UI . 

looking back when checking the commands with NUT I noticed that for enable / disable features I had to use my configured username / password.

After adding that to peaNUT all is ok,

So maybe somewhere in the docs or in the UI certain elements could show a user that he needs the username/password for that.

Brandawg93, 2025-02-14:
This should be fixed in v5.3.0. If you still have any troubles, let me know.

aroundmyroom, 2025-02-15:
@Brandawg93 I like the way it is implemented, but there is no validation of the username and password. I can put in anything what I want but if that is wrong, the system still tells me that it will enable / disable (but actuall can't) .. 
if no username or password is given: no tests . I like that.

Also when managing servers, the test function would assume that you are testing the server with IP, port and username/password (which is not), as the test will be valid as long as it can 'reach?' nut (even without the need for username and password).

I do not need to have this 'fixed' but a 'note' could give some info about what is the usage for username/password etc..

Brandawg93, 2025-02-15:
Good point. Unfortunately this is largely a limitation of the NUT server implementation. It's odd to me that I can input incorrect credentials and it's fine with it until it needs to use them.

But I've made a new PR (#182) that should handle it better from PeaNUT's side. If you have credentials in your server config, it will force check them with the connection test. I'd prefer to do something intuitive to the user as opposed having documentation on the page.

Try it out and let me know if this seems intuitive to you as well. Just use the `test` tag when you pull the docker image.

`brandawg93/peanut:test`

aroundmyroom, 2025-02-16:
I will try to do the test as I am not running docker. Peanut is running here in proxmox as lxc ;)

aroundmyroom, 2025-02-18:
@Brandawg93  tested with dockerge

services:
  peanut:
    image: brandawg93/peanut:test
    container_name: PeaNUT
    restart: unless-stopped
    volumes:
      - /path/to/config:/config
    ports:
      - 9000:9000
    environment:
      - WEB_PORT=9000
networks: {}

But I do not see a difference in usage with username / password. When I do the test the connection is always successful even when the password is fake

I like the text

"Username and password are required for commands"

I wonder if the image is the test image I have .. but ok ..

Brandawg93, 2025-02-18:
That’s probably because I made a test build for a different feature. I’ll revert it and let you know.

Brandawg93, 2025-02-18:
New test is up.

aroundmyroom, 2025-02-18:
I'm testing but when doing the test I get an error 500 so it fails

|  ⨯ Error: Login failed
PeaNUT  |     at i.checkCredentials (.next/server/chunks/5604.js:2:2079)
PeaNUT  |     at async R (.next/server/chunks/5604.js:1:4198) {
PeaNUT  |   digest: '280xxxxx1939'

[image]

username and password are ok.

Brandawg93, 2025-02-18:
Hm. Maybe double check the credentials and where they're configured? When I try on my personal device, it fails with invalid credentials and passes with valid credentials. [Here](https://github.com/Brandawg93/PeaNUT/blob/main/config/upsd.users) is where they are configured for my testing. Username is in brackets.

aroundmyroom, 2025-02-19:
Thanks for the info, I seem to have fixed it

my username and pwd were ok, but  actions were ie in capital like actions = SET and instcmd = ALL
also upsmon primary I did not have. 

modifying and restarting NUT solved that.  

[admin]
        password = nut_test
        actions = set
        actions = fsd
        instcmds = all
        upsmon primary

Case closed ;)

Brandawg93, 2025-02-19:
Awesome! I'll publish this in the next release and then close this.

Brandawg93, 2025-02-20:
This is live in v5.4.2.

closed 2025-02-20 by Brandawg93